mne-tools/mne-python #9078: [ENH] Add OpenNeuro Dataset Compatibility to Examples/Tutorials

alexrockhill, 2021-03-15:
As of now, mne sample data is hosted in OSF through owner privileges. My understanding is that the CIs have to pull and download the sample data every time (if that's wrong then this would probably slow things down) so it might make sense to have a clean API to make OpenNeuro data accessible maybe with nice keywords like `JH_ECoG`.

cc @larsoner, @adam2392

adam2392, 2021-03-15:
Agreed if possible. Then this entire PR: https://github.com/mne-tools/mne-python/pull/9076

would be unnecessary for adding the dataset step at least.

larsoner, 2021-03-15:
> CIs have to pull and download the sample data every time (if that's wrong then this would probably slow things down)

They actually get cached by CircleCI and it's only a couple of minutes (out of a 2h build) to deal with the caching

>  it might make sense to have a clean API to make OpenNeuro data accessible maybe with nice keywords like JH_ECoG.

Indeed it might make sense at some point. We'd want to make sure there is some way to get *subsets* of the data, too, though, since we often can't/won't handle an entire given dataset.

The rate at which we add datasets (and handful per year) and the overhead that this would save (< 10 lines per PR, plus < 30 min? to repackage + upload to OSF if necessary) means that the gains might not be so big for our own internal use. But giving people a nice public interface in MNE to download (parts of) OpenNeuro datasets in general would be pretty cool. It starts to sound a bit like it should be its own small package, though, since things like Nibabel, nilearn, dipy, etc. would probably also be able to use and contribute to it. Then we can require or vendor it in MNE.

alexrockhill, 2021-03-15:
Okay maybe we should finish the PR that uses it first and then see if it should be a priority for this week

adam2392, 2021-03-15:
I remember @hoechenberger having a python api interface for openneuro now that we're discussing it.

hoechenberger, 2021-03-15:
https://github.com/hoechenberger/openneuro-py

alexrockhill, 2021-03-15:
Nice! So could this be a dependency for tutorials and have the data cached and work fast also or is that not something worth pursuing?

sappelhoff, 2021-03-16:
This also came up some years ago in https://github.com/mne-tools/mne-python/issues/6687

I think that one can probably be closed?

hoechenberger, 2021-03-16:
@alexrockhill @adam2392 Ping me tomorrow so we can discuss whether `openneuro-py` supports everything you'll need or if we need to add a few enhancements :)

alexrockhill, 2021-03-23:
Maybe we could table this until the next dataset we'd like to use for an example, that way we might have a better idea what would be nice as far as functionality to be added

hoechenberger, 2021-03-25:
Could you bring me up to speed what's the current status here? Do you need any help from my end? Does `openneuro-py` do everything you need?

alexrockhill, 2021-03-25:
The idea was that it would be nice to be able to use datasets on OpenNeuro for mne examples. It would be nice to only get one subject and one run for speed of execution for example. I think it might help to make an example with a dataset on OpenNeuro even if the theme was just using an OpenNeuro dataset with mne. That might be better on `mne-bids` though. I can start a PR for that now.

agramfort, 2021-03-25:
Let's start with openneuro-py in autoreject to see how it files.

there is a burning need in autoreject that downloads >= 16GB of data
for the example
that in the end uses only one subject

sappelhoff, 2021-03-25:
> there is a burning need in autoreject that downloads >= 16GB of data

:+1: see https://github.com/autoreject/autoreject/issues/201

alexrockhill, 2021-03-25:
I think `openneuro-py` has pretty much everything already needed to include sample data using it. The only things that would be nice to add are documentation and regular expressions instead of just startswith for including files. For the latter, the main use would be including specific runs and I guess you sometimes wouldn't need the sidecar files but they are so small it seems good to include them. Otherwise, this issue might be closed and it's just a matter of adding an example or two to the docs so that this becomes a way to use example data regularly. Those seem like better issues on the `openneuro-py` repo.

closed 2021-03-25 by alexrockhill