mne-tools/mne-python #9093: ENH: Add montage for Artinis OctaMon

HanBnrd, 2021-03-16:
#### Describe the new feature or enhancement
We would like to add a default montage for the Artinis OctaMon fNIRS device. This would enable to have spacial locations useful for some pre-processing functions (such as the modified Beer Lamber law) and having visualization on a brain map.

[image]

#### Describe your proposed implementation
Adding a file in [mne/channels/data/montages/](https://github.com/mne-tools/mne-python/tree/main/mne/channels/data/montages), modifying [`set_montage()`](https://mne.tools/stable/generated/mne.io.Raw.html#mne.io.Raw.set_montage) (in Raw but also other data types) as and probably the loading functions.

#### Additional comments
Not sure what file format is best to use (csd/sfp/txt/elc), would you have recommendations?

HanBnrd, 2021-03-16:
We discussed it with @rob-luke and I would be happy to work on a PR for this if you are okay with it.

agramfort, 2021-03-16:
no objection

is there such a standard montage file available somewhere?

our montage files are shared between packages.

>

HanBnrd, 2021-03-16:
I don't think there is a standard montage file with 3D coordinates, will try to see if I can adapt the montage template the Artinis software has.

agramfort, 2021-03-16:
sounds good

>

rob-luke, 2021-03-16:
@HanBnrd has received the locations in MNI ICBM 152 space. The file format provided by the vendor looks custom. Rather than writing another custom file format reader, I suggest we manually convert these to one of the existing formats perhaps sfp. Would this be acceptable @agramfort @drammock @larsoner

rob-luke, 2021-03-17:
We will need to edit: https://github.com/mne-tools/mne-python/blob/c638f74ac727aeaca3b9616e56115b990a84c7ba/mne/channels/_standard_montage_utils.py#L108

https://github.com/mne-tools/mne-python/blob/c638f74ac727aeaca3b9616e56115b990a84c7ba/mne/channels/montage.py#L1226-L1306

larsoner, 2021-03-17:
Yes that sounds reasonable. If the format is not complicated (e.g., some text file that can be easily parsed by eye) I would just add a quick reader for it rather than convert.

agramfort, 2021-03-17:
ok for me

HanBnrd, 2021-03-17:
> If the format is not complicated (e.g., some text file that can be easily parsed by eye) I would just add a quick reader for it rather than convert.

We were planning to use the elc structure as the original files are quite similar, even though we have optodes and not electrodes:
- [original txt file sent by Artinis](https://github.com/mne-tools/mne-python/compare/main...HanBnrd:artinis-montages#diff-fa76b4fa1b84fdedc992433dec624eeb2ab5566d9b98f9463f704504ef4b9073)
- [the same montage converted into elc](https://github.com/mne-tools/mne-python/compare/main...HanBnrd:artinis-montages#diff-9ba80d9351b39faa894513f96f4c92a7ab509a6878e71072ca217324c53adf1d)

Would that be okay?

agramfort, 2021-03-17:
so far so good

>

PR #9141 (2021-04-23): WIP: Adds support for fNIRS data to set_montage and new standard artinis montage
#### Reference issue
Add montage for Artinis OctaMon and Brite23 Fixes #9093

#### What does this implement/fix?
1. This adds support for fNIRS data to set_montage
2. Add a new standard montage for the artinis brand device as used in @HanBnrd's lab

#### Additional information
This PR has had feature creep and now contains a suggestion for importing data as csv files, to support the reading of unstructured data this PR now contains new code to check the naming of channels, etc. I have suggested reverting this PR b [... cut]
closed 2021-04-23 by larsoner