mne-tools/mne-python #9101: BUG: parse_folder() with n_jobs>1 for mne.Report causes attribute-error

marsipu, 2021-03-16:
#### Describe the bug
When changing the n_jobs-parameter for the parse_folder-method of mne.Report to a value greater than 1, an error is thrown.

#### Steps to reproduce
```
import mne

path = mne.datasets.sample.data_path(verbose=False)
report = mne.Report(verbose=True)
report.parse_folder(path, pattern='*raw.fif', render_bem=False, n_jobs=2)
```

#### Expected results
No error.

#### Actual results
```
Traceback (most recent call last):
  File "<decorator-gen-368>", line 22, in parse_folder
  File "C:\Users\marti\PycharmProjects\mne-python\mne\report.py", line 1620, in parse_folder
    r = parallel(p_fun(self, fname, info, cov, baseline, sfreq, on_error,
  File "C:\Users\marti\PycharmProjects\mne-python\mne\parallel.py", line 128, in run
    return fun(*args, **kwargs)
  File "C:\Users\marti\anaconda3\envs\mnedev\lib\site-packages\joblib\parallel.py", line 1054, in __call__
    self.retrieve()
  File "C:\Users\marti\anaconda3\envs\mnedev\lib\site-packages\joblib\parallel.py", line 933, in retrieve
    self._output.extend(job.get(timeout=self.timeout))
  File "C:\Users\marti\anaconda3\envs\mnedev\lib\site-packages\joblib\_parallel_backends.py", line 542, in wrap_future_result
    return future.result(timeout=timeout)
  File "C:\Users\marti\anaconda3\envs\mnedev\lib\concurrent\futures\_base.py", line 440, in result
    return self.__get_result()
  File "C:\Users\marti\anaconda3\envs\mnedev\lib\concurrent\futures\_base.py", line 389, in __get_result
    raise self._exception
AttributeError: 'Report' object has no attribute 'projs'
```

#### Additional information
The error seems to be coming from line 362 in report.py. The report-object doesn't have the _projs_-attribute as if it wasn't initialized correctly or somehow lost attributes when being passed to __iterate_files_ in line 1620

hoechenberger, 2021-03-16:
I can reproduce this locally, but so far I'm not sure where this is coming from … :(

marsipu, 2021-03-16:
Guillaume and I found a workaround by adding `prefer='threads'` to parallel_func in l. 1619 of report.py.
But this still doesn't explain the disappearing attribute (and threading shouldn't give much of a perfomance boost)

PR #9109 (2021-03-17): BUG: Fixing Report multiprocessing
#### Reference issue
Fixes #9101.

#### What does this implement/fix?
This fixes the bug appearing when calling Report().parse_folder() with n_jobs>1 by adding _projs_ to the list of pickled attributes in _get_state_params().

Solution suggested by @larsoner
closed 2021-03-17 by agramfort