mne-tools/mne-python #9110: make property plot launched from ica.plot_sources(raw,) an option.

bloyl, 2021-03-16:
If you use `ica.plot_sources` with a `raw` instance and then right click on the ch_name you get an `mne.viz.plot_ica_properties` plot where fixed length epochs have been generate from the raw. This takes a long time and isn't tremendously meaningful for raw data.

In my use case I'm only checking for artifacts (EOG/ECG/ etc) so what I want to see is the just the topoplot (note this is the old behavior).

would it be possible to optionally choose the new (`plot_ica_properties`) or the old (just topomap) behavior?

drammock, 2021-03-16:
@bloyl do you mean to automatically triage which plot we generate based on the instance type? Or do you mean let the user pick when they instantiate the plot?

bloyl, 2021-03-16:
I don't have a preference.

I think for raw instances id like to be able to get the just topoplot
behavior. I can't think of when I would use the plot_ica_propeties with a
raw instance. For epochs the plot_ica_properties is probably what I want.

If noone wants the plot_ica_properties behavior for raw. I'd be fine with
the triage of
Raw-> plot_topo
Epochs-> plot_ica_properties

If we want both behaviors available for raw we'd obviously need a user
option.

drammock, 2021-03-16:
OK, let's go with 

> Raw-> plot_topo
> Epochs-> plot_ica_properties

bloyl, 2021-03-16:
Sounds good.

drammock, 2021-03-17:
assigned to @vagechirkov and @anaradanovic

kimcoco, 2021-03-17:
I was actually super happy to have the plot_properties come up when right clicking on raw ica sources with the latest version. We need this in our lab, where we reject noise components with the ica on reference channels method. The properties plot is here the best, and more informative about whether sth is noise than just a topo plot.
I also don't know why it would be wise to have different behavior for epochs or raw.

kimcoco, 2021-03-17:
I know, but I mean exactly the right-click behavior in the interactive interface. I have to do a lot of property plots for each raw file, looking at several components. It made it so much easier just right-clicking on them, instead of having to call a seperate function and to pass the right component number or a list of them.

PR #9183 (2021-03-24): ENH: Add fast plotting ICA properties
closes #9110 

@larsoner [suggestion](https://github.com/mne-tools/mne-python/pull/9134#issuecomment-803042712) in the PR #9134:

- precompute epochs ICA sources only once 

 Line profiler output for `_plot_ica_properties` and `_fast_plot_ica_properties`:

**Sample data**
<details>

```
<Raw | sample_audvis_raw.fif, 376 x 166800 (277.7 s), ~481.8 MB, data loaded> - sfreq = 600.614990234375

Total time: 0.539739 s
File: /mne-python/mne/viz/ica.py
Function: _plot_ica_properties at line 122

Line #      Hits         Time  Per Hit   % Time  Line Contents
===== [... cut]
closed 2021-03-24 by drammock