mne-tools/mne-python #9111: Add montage.add_estimated_fiducials

larsoner, 2021-03-16:
New method of `montage` that, if the montage is in `MRI` / FreeSurfer surface RAS coordinates, will estimate the fiducials (using mne.coreg.get_mni_fiducials) and add them to the montage. For example this docstring:
```
"""Estimate fiducials based on FreeSurfer parameters

Parameters
---------
subject : ...
subjects_dir ...
```
It should help iEEG folks get their montages to be MNE/FIF compatible without requiring a bunch of changes in MNE. For example they would have:
```
montage = ...  # wherever they get their electrodes in MRI (FS Surface RAS) coords
montage.add_estimated_fiducials(subject, subjects_dir)  # adds LPA/Nasion/RPA based on FS files
trans = montage.get_native_head_t()  # head<->MRI trans can now be computed properly
```
Then when they `set_montage`, MNE internally converts to all electrode locations to head coords as usual (for now at least), and iEEG folks pass `trans` to functions that need them (like `plot_alignment`) and now all coordinate frames are properly defined. Still future compatible with the dream of MNE supporting coordinate frames other than head for EEG electrodes (since it just allows us to properly define the Neuromag coordinate frame), but gives us a bit more leeway to punt on that for now.

Discussed with @libertyh as a possible partial solution for iEEG folks.

adam2392, 2021-03-16:
I can do this really quickly. 

Does this function just call: `mne.coreg.get_mni_fiducials`? 

Is there a unit-test in mind we can apply here?

larsoner, 2021-03-16:
Excellent!

> Does this function just call: mne.coreg.get_mni_fiducials?

Yes, and adds it to the montage in the right way (whatever that is).

> Is there a unit-test in mind we can apply here?

You can test it on fsaverage, it should match (to numerical precision) the built-in fsaverage trans .fif file we have. You can test it on sample, and it should match to some correlation limit (0.98+ I'm guesssing) the manually-created trans for `sample`.

PR #9118 (2021-03-18): [MRG; ENH] Add "add_estimated_fiducials" to DigMontage
#### Reference issue
Closes: #9111 

Will enhance: #9087 and `plot_seeg` related tutorials in the future. Necessary ingredients for dataset though:

- FreeSurfer subject-specific files
- electrode coordinates in FS subject-specific T1w

#### What does this implement/fix?
Add `add_estimated_fiducials` function to `DigMontage`.

Implements unit test... todo

#### Additional information
From BIDS convos of mne sprint chat.
closed 2021-03-18 by larsoner