mne-tools/mne-python #9112: ENH: IDE configuration docs

drammock, 2021-03-16:
for future code sprints / general friendliness to new contributors, it might be nice to have a website page that steps through the various IDE settings that are helpful when contributing to MNE-Python.  Examples:

- how to add automatic style checking to the IDE
- how to add a ruler at 79 characters to the editor

It might also include other "tips and tricks" like how to properly link a PR to an issue that it fixes, by using GitHub's magic phrases (e.g., explain that `closes #XXXX` will work, but `solves issue #XXXX` will not)

drammock, 2021-03-16:
some examples:

#### spyder

- preferences > completion and linting > linting > enable basic linting
- preferences > completion and linting > introspection > advanced > add "mne" to list of modules to preload for code completion
- preferences > editor > show vertical lines at == 79
- preferences > editor > advanced settings > docstring type == Numpy

sappelhoff, 2021-03-16:
:+1:  and removing trailing whitespace 

potentially even handling the GitHub inline reviews, committing these suggestions from the web UI (single commits, or batches)

hoechenberger, 2021-03-21:
Partially related, but a while a go I started documenting how to get the basics of a Python & JS development environment set up on macOS:

https://docs.google.com/document/d/1ONwEPWYYBpuAhlVEIv5gZytGPC4i8B4uxixTLMcczF0/edit?usp=sharing

rkalahasty, 2021-12-06:
Could I potentially try and tackle this?

drammock, 2021-12-06:
> Could I potentially try and tackle this?

sure, feel free to open a PR. The file should probably live at `doc/install/configure_ide.rst`

rkalahasty, 2021-12-06:
Alright, Ive opened a PR creating the file.