mne-tools/mne-python #9116: Add SSD to the Decoding (MVPA) tutorials

vpeterson, 2021-03-16:
#### Proposed documentation enhancement
Add an example of how SSD can be used within a machine-learning pipeline. At the moment we have an [example](https://mne.tools/stable/auto_examples/decoding/plot_ssd_spatial_filters.html#sphx-glr-auto-examples-decoding-plot-ssd-spatial-filters-py) on how SSD enhances oscillatory brain activity, but not how it can be integrated within a decoding pipeline. The examples can then be added in the `tutorials/machine-learning/plot_sensors_decoding.py`, which are the moment SSD is missing there. 

My idea is to show how SSD can be beneficial before applying CSP or SPoC. A brief explanation on the formulation of SSD can be added too, as done for CSP. 

What do you think  @dengemann @larsoner @sappelhoff @drammock ??

drammock, 2021-03-16:
sounds like a good plan.  If you make that change, will we still need the example anymore?  Or will the information be different enough that it makes sense to have 2 separate files?

vpeterson, 2021-03-17:
Well, I might check that, since not sure if the data we used for that example will be suitable to show the effect of SSD before decoding. 

> sounds like a good plan. If you make that change, will we still need the example anymore? Or will the information be different enough that it makes sense to have 2 separate files?