mne-tools/mne-python #9120: FIX, ENH, DOC:  First_samp support for SourceSimulator class

NeuroLaunch, 2021-03-17:
#### Describe the new feature or enhancement
Currently, simulating raw data with a `SourceSimulator `object (in _simulation/raw.py : simulate_raw()_) causes extra time points at the beginning if the source events are offset in time, even when a `first_samp` argument is included to nominally account for that offset.  This could happen, for example, if an experimental raw object and associated event list are used to instantiate _simulate_raw()_.  Effectively, the `first_samp` argument only changes the simulated raw object's absolute timing attributes (.first_samp, .last_samp, .first_time, etc), but doesn't affect how sources are actually added to it.  This means that if the first event has an arbitrarily large time stamp, then the simulator object (and therefore the resulting raw object) will be arbitarily long.

The documentation should also be updated.  For example, presently there is no indication that a `SourceSimulator `object is a valid argument for the _simulate_raw()_ constructor.

#### Describe your proposed implementation
I propose to add `first_samp `support to the `SourceSimulator `class, so that event waveforms added to an instant of the class (via _.add_data(_)) are appropriately time stamped.  This can be implemented by simply subtracting the `first_samp ` value from the integer event times (which are expressed as absolute samples) when adding event data.  The `first_samp `value would be passed as an argument at the time the source object is instantiated; it would default to 0 (which would not influence existing code) and event times less than the value would cause an error.

I am inclined to assert that the `first_samp` argument for _simulate_raw()_ be the same as the `first_samp` attribute of the simulator object.  In any case, a suitable test function should make sure that added events in the raw object occur at the intended time samples.  (Presently there is no test for using source simulators with _simulate_raw()_.)

The alternative way to simulate raw data with _simulate_raw()_ is with a `SourceEstimate `object or an iterable of `SourceEstimate `objects.  The `first_samp` argument shouldn't affect how the raw data is created with this approach, as the STCs are simply concatenated in time.

#### Describe possible alternatives
Rather than changing the SS class definition, MNE users could be made aware of the current `first_samp` limitations via an instructive example.

#### Additional comments
With @cmista for Code Sprint 2021; I have mentioned the proposed change to @larsoner

larsoner, 2021-03-17:
> I propose to add first_samp support to the SourceSimulator class

+1

> I am inclined to assert that the first_samp argument for simulate_raw() be the same as the first_samp attribute of the simulator object.

Sure, I'd consider this a bug fix for people.

agramfort, 2021-03-17:
sounds good !

>

PR #9166 (2021-03-25): MRG, FIX, ENH: First_samp support and simulation tests
#### Reference issue
Fixes #9120.

#### What does this implement/fix?
Modified _mne.simulation.raw.simulate_raw()_ and _mne.simulation.source.SourceSimulator_ to properly interpret `first_samp `parameter.  Adding events to a `SourceSimulator` would not account for any sample offsets, such that the resulting raw data created with _simulate_raw()_ would have shifted event-aligned waveforms regardless of the `first_samp `argument.

#### Additional information
In order to properly effect the sample offset change, I reinterpreted the `duration` [... cut]
closed 2021-03-25 by hoechenberger