mne-tools/mne-python #9125: AverageTFR.plot() only plots one channel

cbrnr, 2021-03-17:
Although [`AverageTFR.plot()`](https://mne.tools/stable/generated/mne.time_frequency.AverageTFR.html?highlight=averagetfr#mne.time_frequency.AverageTFR.plot) has a `picks` argument, only the first channel is plotted. I don't think this is by design - or is it? If so, then at least the docstring is confusing, because it explicitly mentions that `None` selects all good channels and `"all"` selects all channels.

```python
import numpy as np
import mne

n_chans = 10
n_epochs = 100
tmin, tmax = -2, 5
fs = 256
info = mne.create_info(n_chans, fs, "eeg")
data = np.random.rand(n_epochs, n_chans, (abs(tmin) + abs(tmax)) * fs + 1)
epochs = mne.EpochsArray(data, info, tmin=tmin)

freqs = np.arange(1, 51)

tfr = mne.time_frequency.tfr_multitaper(epochs, freqs=freqs, n_cycles=freqs,
                                        average=False, return_itc=False)
tfr.average().plot()
```

agramfort, 2021-03-17:
probably a bug...

>

cbrnr, 2021-03-18:
I think it plots the average of picked channels, even though `combine=None` is the default (and which apparently doesn't work). So unless I'm using it totally wrong I also think this is a bug. @mmagnuski you seem to have used TFRs a lot, WDYT?

eort, 2021-03-18:
If I am not mistaken, this should essentially describe the same bug as #8013 
I committed a fix that is currently waiting to be checked (#9150). Perhaps you, @cbrnr can also check whether this proposed changed solve your problems?

Btw, the current default behaviour for `picks=None` is plotting all channels in a separate figure. So unless you reduced your tfr object to only the channels your are interested in, you better set `combine` to `mean` or `rms`, otherwise you end up with *a lot of* figures.

cbrnr, 2021-03-18:
I only get exactly one figure with `picks=None` (which is the default behavior).

eort, 2021-03-18:
Yes, that's right. I meant in my commit. Fixing that issue of always getting exactly one figure, wIthout changing the default behaviour of picks, will lead to as many figures as you have channels.

mmagnuski, 2021-08-02:
Seems to have been fixed by #9150.

closed 2021-08-02 by mmagnuski