mne-tools/mne-python #9186: Reading EDF file yields incorrect scaling

hoechenberger, 2021-03-23:
Reported on the forum:
https://mne.discourse.group/t/eeg-data-in-edf-not-converting-in-volt/

The user provided an EDF file which specifies in its header that the data are in µV. When reading this file with MNE-Python, the data is treated as V.

cbrnr, 2021-03-23:
Opening the file linked in the discussion thread in a hex editor shows that the units are declared as `.V` - so this is clearly an invalid EDF file.

drammock, 2021-03-23:
invalid file → `wontfix`?

cbrnr, 2021-03-23:
Yes. MNE has never been super strict, but we should not try to interpret `.V` as `µV`.

cbrnr, 2021-03-23:
Which raises the question: since we do load the file, should we perhaps be stricter and raise an error? I've always argued not to do that, because having incorrectly scaled data is better than having no data.

larsoner, 2021-03-23:
I agree, +1 for a warning when we can't safely interpret the units. Then the user can fix their file or adjust their processing as necessary

hoechenberger, 2021-03-23:
> Which raises the question: since we do load the file, should we perhaps be stricter and raise an error? I've always argued not to do that, because having incorrectly scaled data is better than having no data.

My personal feeling is we should throw an error. I feel that incorrectly scaled data is **worse** than no data ;)

hoechenberger, 2021-03-23:
Ok, a warning it is!

Sumalyo, 2021-03-23:
I have tried to address the issue, showing a warning if the unit read is in Volts

PR #9187 (2021-03-25): MRG: Support µV in EDF files (in addition to uV)
This shows a warning if edf file data is given in Volts as mentioned in the issue discussions. Though it interprets the data in Volts, it informs the data about a possible problem in scaling. Fixes #9186.
closed 2021-03-25 by agramfort