mne-tools/mne-python #9193: Crashes when using read_epochs_eeglab on files with only one trial.

jackz314, 2021-03-25:
#### Describe the bug
When reading in single trial EEGLAB files as epochs, MNE will crash when initializing `EpochsEEGLAB`. This is because the trial count is 1, so creating the `event_id` dict is skipped, and when [this line](https://github.com/mne-tools/mne-python/blob/13a0de51bd0f13113d866e73522fb60bdde37809/mne/io/eeglab/eeglab.py#L507) references `event_id`, an `AttributeError` will be thrown. Like this one:

```
test_epochs.py:3046: in test_save_set
    epochs_read = read_epochs_eeglab(temp_fname)
../io/eeglab/eeglab.py:275: in read_epochs_eeglab
    epochs = EpochsEEGLAB(input_fname=input_fname, events=events, eog=eog,
<decorator-gen-240>:24: in __init__
    ???
../io/eeglab/eeglab.py:507: in __init__
    for key, val in event_id.items():
E   AttributeError: 'NoneType' object has no attribute 'items'
```

#### Steps to reproduce
I discovered this when writing tests for my PR #9192, I was following another similar test [`test_save_complex_data` ](https://github.com/mne-tools/mne-python/blob/13a0de51bd0f13113d866e73522fb60bdde37809/mne/tests/test_epochs.py#L3017) that also saves single-trial epochs to a file and reads it back as a test. I simply wrote the same data into an EEGLAB set file and read it back. Alternatively, you can also try `read_epochs_eeglab` on any single-trial set file.

#### Expected results
Since single-trial epochs aren't really useful or meaningful (from my understanding so far), I understand if this isn't supported, but some better clarifications on this would probably make more sense to other people, for example, throwing another exception specifically for single-trial data when attempting `read_epochs`. 

#### Actual results
N/A

#### Additional information
Platform:      Linux-5.8.0-45-generic-x86_64-with-glibc2.32
Python:        3.8.6 (default, Jan 27 2021, 15:42:20)  [GCC 10.2.0]
Executable:    /usr/bin/python
CPU:           x86_64: 8 cores
Memory:        31.3 GB

mne:           0.23.dev0
numpy:         1.19.5 {blas=openblas, lapack=openblas}
scipy:         1.4.1
matplotlib:    3.3.4 {backend=Qt5Agg}

sklearn:       0.24.1
numba:         0.52.0
nibabel:       Not found
nilearn:       Not found
dipy:          Not found
cupy:          8.5.0
pandas:        1.1.4
mayavi:        Not found
pyvista:       Not found
vtk:           Not found

agramfort, 2021-03-25:
to me it's a bug

can you open a PR to also support single epoch files?

>

jackz314, 2021-03-27:
Opened PR at #9204.

drammock, 2021-03-27:
Hi @jackz314 thanks for opening a PR to fix this! FYI our preferred workflow is to close issues when the associated PR is *merged* rather than *opened* (ideally by using GitHub magic phrase "closes #XXXX" in the PR description).

jackz314, 2021-04-05:
Fixed in #9204

closed 2021-04-05 by jackz314