mne-tools/mne-python #9195: Limit Epochs rejection to subset of channels / exclude channels

hoechenberger, 2021-03-25:
I'm working with EEG recordings that include data from electrodes places very close to the eyes, capturing lots of EEG activity. I use these channels to identify and  auto-clean EOG artifacts using ICA. I have ICA operate on Epochs.

Because I want to avoid passing any Epochs with spuriously large signal amplitudes to ICA, I set a `reject` parameter for my epochs. This successfully removes a bunch of epochs where the signal indeed jumped way out of bounds for an electrode or two.

However, the EOG amplitudes I get in Fp1 and Fp2 are so big that in order to not remote a large number of epochs based on the EOG signal alone, I'd have to set the rejection threshold so high that I'd then be unable to reject epochs where some of the other electrodes went rogue. So this is a problem for me: either I set the threshold so high that it more or less becomes useless; or I lower it and sacrifice a lot of data due to the EOG signal.

What I'd like to do is specify "drop epochs as bad if amplitude exceeds X, unless this occurs in Fp1 and Fp2".

Wanted to discuss with you

- if you think this would be good to have (I very much do!)
- and in a second step, what the API could look like

Thanks!

cc @cbrnr @agramfort

cbrnr, 2021-03-25:
I think this is a good idea. We could pass a `dict` such that it allows you to specify channel-specific thresholds, e.g. `reject={"Fp1": 2e-5, "Fp2": 2e-5}`. Not sure how to handle the other channels then though.

hoechenberger, 2021-03-25:
> We could pass a `dict` such that it allows you to specify channel-specific thresholds, e.g. `reject={"Fp1": 2e-5, "Fp2": 2e-5}`. Not sure how to handle the other channels then though.

I like this idea. We could allow for something like:

```
reject={"eeg": 100e-6, "Fp1": 2e-5, "Fp2": 2e-5}
```
and the `Fp1` and `Fp2` specifications would supersede the broader specification of `eeg`

cbrnr, 2021-03-25:
The only problem I see is if there is a channel called "eeg".

hoechenberger, 2021-03-25:
Or we just go with the ability to exclude channels from rejection?

```
reject={"eeg": 100e-6}, reject_exclude=["Fp1", "Fp2"]
```

cbrnr, 2021-03-25:
But this is different from channel-specific thresholds. I'm not sure what would be more useful.

hoechenberger, 2021-03-25:
> But this is different from channel-specific thresholds. I'm not sure what would be more useful.

yes but you could apply channel-specific thresholds that way too by rejecting repeatedly with different settings, it would just be a little ugly. ;)

cbrnr, 2021-03-25:
😆 I agree, this would be ugly.

larsoner, 2021-03-25:
> and the Fp1 and Fp2 specifications would supersede the broader specification of eeg

Agreed on this API and the other case:
```
reject={"Fp1": 2e-5, "Fp2": 2e-5}
```
would mean "don't reject based on EEG for anything other than Fp1 and Fp2 at these levels", i.e., acts the same way as
```
reject={"eeg": np.inf, "Fp1": 2e-5, "Fp2": 2e-5}
```

> The only problem I see is if there is a channel called "eeg".

We already have problems in principle with this for `picks` / `_picks_to_idx` and in practice nobody has complained, so I think we can safely ignore this corner case.

From an I/O standpoint we currently just `json.dumps` our `reject` dict, so as long as we preserve the values / prominence in `epochs.reject` dict properly, I think it will "just work" on I/O round-trip, too.

hoechenberger, 2021-03-25:
Ok, everybody cool with going with 
```
reject={"eeg": 100e-6, "Fp1": 2e-5, "Fp2": 2e-5}
```
then..?

drammock, 2021-03-25:
thinking about your particular use case @hoechenberger, would it work to mark the frontal channels as "bad", run the epoching, then set them back to "good" before running ICA?

hoechenberger, 2021-03-25:
@drammock this is a very interesting idea. Yes this could work indeed. I will play with this and see how far I get!

hoechenberger, 2021-04-22:
I implemented @drammock's suggestion in the MNE BIDS Pipeline, but had a followup discussion with @agramfort and we decided to take a different approach after all, namely applying different `reject` parameters before and after ICA, possibly using `autoreject` to help us find sensible thresholds automatically. In any case, I believe this issue can be closed.

closed 2021-04-22 by hoechenberger