mne-tools/mne-python #9198: linear regression

virvw, 2021-03-26:
### Describe the bug
linear_regressions
developper
worked fine a few days ago
no mne update since
now bug
#### Steps to reproduce
  pred_vars = ['TOJ - SJ', 'N_1', 'intercept']
  design = epos.metadata[['task', 'N_1']].copy()
  design['TOJ - SJ']  = np.where(design['task'] == 'TOJ', 1, -1)
  design['intercept'] = 1
  design = design[pred_vars]
  reg = linear_regression(epos, design_matrix=design, names=pred_vars)
#### Expected results
outcomes of reg
#### Actual results
-----------------------------------------------------------------------------------------------------------
Fitting linear model to epochs, (214506 targets, 3 regressors)
Traceback (most recent call last):
  File "<ipython-input-93-7be722fdd449>", line 1, in <module>
    reg = linear_regression(epos, design_matrix=design, names=pred_vars)
  File "C:\Users\vv221713\Anaconda3\envs\mne\lib\site-packages\mne\stats\regression.py", line 85, in linear_regression
    lm_params = _fit_lm(data, design_matrix, names)
  File "C:\Users\vv221713\Anaconda3\envs\mne\lib\site-packages\mne\stats\regression.py", line 118, in _fit_lm
    betas, resid_sum_squares, _, _ = linalg.lstsq(a=design_matrix, b=y)
  File "C:\Users\vv221713\Anaconda3\envs\mne\lib\site-packages\scipy\linalg\basic.py", line 1155, in lstsq
    a1 = _asarray_validated(a, check_finite=check_finite)
  File "C:\Users\vv221713\Anaconda3\envs\mne\lib\site-packages\scipy\_lib\_util.py", line 266, in _asarray_validated
    raise ValueError('object arrays are not supported')
ValueError: object arrays are not supported
-----------------------------------------------------------------------------------------------------------
#### Additional information
windows and linux
win
(mne) C:\Users\vv221713>python -c "import mne; mne.sys_info()"
Platform:      Windows-10-10.0.18362-SP0
Python:        3.8.5 (default, Sep  3 2020, 21:29:08) [MSC v.1916 64 bit (AMD64)]
Executable:    C:\Users\vv221713\Anaconda3\envs\mne\python.exe
CPU:           Intel64 Family 6 Model 142 Stepping 12, GenuineIntel: 8 cores
Memory:        15.6 GB
mne:           0.22.dev0
numpy:         1.19.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.5.2
matplotlib:    3.3.2 {backend=Qt5Agg}
sklearn:       0.23.2
numba:         0.51.2
nibabel:       3.2.0
cupy:          Not found
pandas:        1.1.3
dipy:          1.3.0
mayavi:        4.7.2
pyvista:       0.27.2 {pyvistaqt=0.2.0, OpenGL 4.5.0 - Build 26.20.100.7755 via Intel(R) UHD Graphics}
vtk:           9.0.1
PyQt5:         5.15.1
linux
Python:        3.8.5 (default, Sep  3 2020, 21:29:08) [MSC v.1916 64 bit (AMD64)]
Executable:    C:\Users\vv221713\Anaconda3\envs\mne\python.exe
CPU:           Intel64 Family 6 Model 142 Stepping 12, GenuineIntel: 8 cores
Memory:        15.6 GB
mne:           0.22.dev0
numpy:         1.19.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.5.2
matplotlib:    3.3.2 {backend=Qt5Agg}
sklearn:       0.23.2
numba:         0.51.2
nibabel:       3.2.0
cupy:          Not found
pandas:        1.1.3
dipy:          1.3.0
mayavi:        4.7.2 (edited)

hoechenberger, 2021-03-26:
For those following this issue:
@virvw will send me the epochs in question next week so I can explore the problem

larsoner, 2021-03-26:
I have a feeling it's this line:
```
  design['TOJ - SJ']  = np.where(design['task'] == 'TOJ', 1, -1)
```
`np.where` will *always* return a tuple of ndarray, so you probably want:
```
  design['TOJ - SJ']  = np.where(design['task'] == 'TOJ', 1, -1)[0]
```
to get a 1D array rather than a 2D array (what your original code will produce). I'm guessing that later in the code there is some `np.array([x, y])` call where `x` is 2D and `y` is 1D, yielding an object array.

mmagnuski, 2021-03-26:
The error `ValueError: object arrays are not supported` suggests that the design matrix is of object dtype.
First, according to the documentation the `design_matrix` should be an array, while it seems to be a dataframe in your example. Then, before converting to numpy array, make sure that `design.dtypes` does not give you 'object' dtype for any of the columns.

larsoner, 2021-03-26:
oh wait, you're not using `np.where` to return indices, you're using values. Never mind, that's not going to be the problem!