mne-tools/mne-python #9199: Per channel annotation is too slow

mehdikuchi, 2021-03-26:
#### Per channel annotation with clear visual appearance 
The Per channel annotation already developed in the dev version of the MNE, highlights the whole span of the data under view. This makes  event localization visually too difficult. 

#### What I have tried: 

*I have changed the mne/viz/_figure.py code so that I could get the following annotation per channel: 
[image]
The problem is scanning the window becomes too slow. 
*

Is there anything that can be added to speed up the scanning of the data?

larsoner, 2021-03-26:
> Is there anything that can be added to speed up the scanning of the data?

Is it faster if you don't add the text objects? If so, we could make it so that the text just shows up on a mouseover for example

drammock, 2021-03-26:
it is on our roadmap to have per-channel annotations show up as highlighted regions (rather than text) which I think will be faster to draw.  cf #8946.  It's milestoned for the next release so hopefully you won't have to wait too long for it.

mehdikuchi, 2021-03-29:
Thanks for your replies. 
That is good news. So I wait for the new release.

larsoner, 2023-10-01:
#8946 landed so hopefully this was fixed!

closed 2023-10-01 by larsoner