mne-tools/mne-python #9205: ENH/BUG: Adding an include/exclude bad channels parameter to plot_evoked_topo

ramkpari, 2021-03-27:
Was working on `plot_evoked_topo` ,about issue #8714 with @jasmainak when I noticed that bad channels were influencing ylim calculations when no ylim parameters were specified during the function call.  Currently , The plots from `plot_evoked_topo` are automatically set to not plot any of the bad channels and also there is no direct way to force `plot_evoked_topo`  to include the bad channels without removing the channels from `info['bads']` or with picks. 

One thing that we were thinking while fixing the bug was to maybe add a include/exclude parameter to the function to explicitly include or exclude the bad channels in the plot that it produces.  Thoughts? 

@larsoner @agramfort @sappelhoff 

#### Steps to reproduce
```python
import numpy as np

import mne

from mne.datasets import sample

data_path = sample.data_path()

fname = data_path + '/MEG/sample/sample_audvis-ave.fif'

# Reading evoked data
condition = 'Left Auditory'
evoked = mne.read_evokeds(fname, condition=condition, baseline=(None, 0),
                          proj=False)

evoked.plot_topo()
```

agramfort, 2021-03-27:
I think that evoked are small so doing something like

evoked.pick(...).plot_topo()

should be quite efficient and avoids complexifying plot_topo API.

regarding plotting bads maybe we plot them but in low contrast mode?

>

jasmainak, 2021-03-28:
`exclude` param is part of `evoked.plot()` ... I'd say there should be some way to plot bad channels.

also `ylim` being calculated from a channel not plotted seems to be a bug to me

ramkpari, 2021-04-07:
A good middle ground option might be to have a `bads` parameter, that decides if the channels marked as bad should be plotted, With `True` plotting bad channels and marking them a specific colour and `False` excluding the bad channels from the plot which is also the current default behaviour and default for the proposed parameter

ramkpari, 2021-04-08:
Thoughts @jasmainak @agramfort

jasmainak, 2021-04-09:
I'd stick to `exclude` as it's more consistent with rest of API

agramfort, 2021-04-10:
ok then to add an exclude param

>

PR #9278 (2021-04-15): MRG ENH/BUG Add exclude parameter to plot_evoked_topo
#### Reference issue
Fixes #9205 .

#### What does this implement/fix?
Adds an exclude parameter to `plot_evoked_topo`.

With the added exclude parameter, ylim's calculations are no longer affected by bad channels that aren't included in the actual plot.
closed 2021-04-15 by agramfort