mne-tools/mne-python #9206: Adding automatic ICA labeling capability ported from EEGLab

adam2392, 2021-03-27:
They have this nice feature which plugs into a database of features I think that can automatically estimate ICA labels in case you have 100s of subjects and can't feasibly run manual ICA, they actually have a cool feature that will match it against known labels to try to estimate if it's eye blink, heart beat, etc. 

_Originally posted by @adam2392 in https://github.com/mne-tools/mne-python/issues/9192#issuecomment-808789978_

drammock, 2021-03-27:
we already have ICA component selection via template matching: https://mne.tools/dev/auto_tutorials/preprocessing/plot_40_artifact_correction_ica.html#selecting-ica-components-using-template-matching

Is the missing piece then that in EEGLAB there's a built-in library of templates, whereas here you have to provide the template from one of your subjects?

adam2392, 2021-03-27:
Yes, so apparently, they have this `ICLabel` database, such that you don't need to provide a template from one of your own subjects: https://github.com/sccn/ICLabel

The idea being:

1. run ICA
2. label component based on ICLabel classifier
3. filter out components
4. re-save "cleaned" data

hoechenberger, 2021-03-27:
ICLabel extracts two to three features:

- ICA topography
- PSD of the recorded data
- optionally autocorrelation

which are then passed to a classifier. The classifier is a [MatConvNet](https://www.vlfeat.org/matconvnet/) CNN. The CNN produces labels for the given features.

It seems that MatConvNet models can be imported as PyTorch nets via [pytorch-mcn](https://github.com/albanie/pytorch-mcn), although this project seems to be inactive.

hoechenberger, 2021-03-27:
Just for the sake of completeness, this file contains their CNN:
https://github.com/sccn/ICLabel/blob/0a2c1454b8870294c36efd6a342169eedbbafb21/netICL.mat

I just tried loading it via pytorch-mcn, but get a NumPy indexing error.

agramfort, 2021-03-28:
@l-omar-chehab worked with me on this a couple of years ago but the project was never finished.
What is critical is to get a good database that reflects the diversity of datasets. If you have only EEG with 30 to 60 channels
it's a lot easier than if you want a system that works with EEG, MEG VectorView, MEG CTF etc. Having a method that works reliably across systems still remains to be developed AFAIK. If you have some spare cycles to work on this let us know with @l-omar-chehab so we share with you what we have.

adam2392, 2021-03-28:
CC @pmyers16 

> @l-omar-chehab worked with me on this a couple of years ago but the project was never finished.
> What is critical is to get a good database that reflects the diversity of datasets. If you have only [...]

Okay, I potentially can help work on this depending on scope of work needed to be done. Yeah I think ideally I just want something that works robustly with scalp EEG 15-100 electrodes and can auto detect eye blink and heart beat (bonus muscle, but idk what that looks like :p) relatively robustly.

Re database: I'm assuming we can't just leverage ICLabel? How would we get such a database?

hoechenberger, 2021-03-28:
> Re database: I'm assuming we can't just leverage ICLabel?

No, since they only seem to share the NN, not its training data. Probably for good reasons (privacy etc.)

> How would we get such a database?

We make our own. We process publicly available data from OpenNeuro, and publish our own datasets. We run ICA on these data and share our "expert opinions" on the decompositions.

We'd need to get multiple expert opinions per dataset to find the consensus. Or even ask the same experts multiple times to apply different decision thresholds ("be lenient", "be strict"). Or to provide a confidence rating.

But this should all be quite simple. We could, in fact, create a small web app that presents experts with select ICA decompositions and collects their responses. Like a survey. This would actually be fun.

drammock, 2021-03-28:
> We could, in fact, create a small web app that presents experts with select ICA decompositions and collects their responses.

Sounds like `braindr` which I think is open source... Might save some time to start with that and swap images of ICA comps in for the MRIs rather than building from scratch.

hoechenberger, 2021-03-28:
> Sounds like `braindr` which I think is open source...

Yeah something like that. Tinder for ICA actually sounds like a lot of fun

agramfort, 2021-03-28:
that was what we started to do 2 years ago too
https://swipe4ica.github.io/#/ :)

happy to resurrect this if you are interested

>

adam2392, 2021-03-28:
So it looks like there is then already a solid amount of data "labeled" in the backend of swip4ica ?

What work is needed at this point to make this fully "functional" in MNE-Python?

agramfort, 2021-03-28:
I would say extract the data from the database, train and validate an ML
model, evaluate on a new
dataset. Then do the engineering effort to package the thing...

>

adam2392, 2021-03-28:
Re database: Is the database well structured and organized and easily accessible? I can run this by a few colleagues to see if they are able to do this w/ my help.

Re model: Sure and then we can store the pytorch weights and have the automatic labeling API require `pytorch` as an optional dependency?

Re new dataset: Should we just split database into training (train and testing sets in K-fold cross validation) and a held-out validation set?

Re engineering effort: I guess then this would entail: i) robust code to re-train model once a critical threshold of new data is labeled in the swipe4ica backend, ii) re-evaluate model, and iii) re-deploy that model weights into MNE?

hoechenberger, 2021-03-28:
> Re model: Sure and then we can store the pytorch weights and have the automatic labeling API require `pytorch` as an optional dependency?

… I'd prefer Keras but that might just be me …
(having a difficult time with PyTorch!)

agramfort, 2021-03-28:
>
> Re database: Is the database well structured and organized and easily
> accessible? I can run this by a few colleagues to see if they are able to
> do this w/ my help.
>

yes it should be

> Re model: Sure and then we can store the pytorch weights and have the
> automatic labeling API require pytorch as an optional dependency?
>
our experiments suggested a random forest was more reliable.

> Re new dataset: Should we just split database into training (train and
> testing sets in K-fold cross validation) and a held-out validation set?
>
yes something like this.

> Re engineering effort: I guess then this would entail: i) robust code to
> re-train model once a critical threshold of new data is labeled in the
> swipe4ica backend, ii) re-evaluate model, and iii) re-deploy that model
> weights into MNE?
>
if we start doing this we need to aim to complete this task. I consider
this to be 2 months of full time job to make something
serious here.

hoechenberger, 2021-03-28:
> if we start doing this we need to aim to complete this task. I consider
> this to be 2 months of full time job to make something
> serious here.

Actually this sounds like an interesting Master's project, no?

agramfort, 2021-03-28:
yes exactly. I was started as 2 master projects (one on the UI / web
and one on the ML validation).

adam2392, 2021-03-28:
> Re database: Is the database well structured and organized and easily accessible? I can run this by a few colleagues to see if they are able to do this w/ my help.
> yes it should be

Awesome! Can you direct me to this and I can scope out how many MS/ugrads I can find to do this? One of my thesis advisors has a course dedicated to basically making open source software in data science; I can ask him if we can leverage a team dedicated to this effort?

> Re model: Sure and then we can store the pytorch weights and have the automatic labeling API require pytorch as an optional dependency?
> our experiments suggested a random forest was more reliable.

Sure yes, I think we should definitely do a fixed K-fold validation where a multitude of classifiers are evaluated for: i) performance (AUC/acc/etc.), ii) model-size and iii) model run-time(?). We pick the best one, and perhaps even expose multiple models depending on a trade-off between FPR and TPR perhaps.

> Re engineering effort: I guess then this would entail: i) robust code to re-train model once a critical threshold of new data is labeled in the swipe4ica backend, ii) re-evaluate model, and iii) [...]

Probably can't do full time w/ anyone here, but I can imagine a team over the course of 3 months can make some headway...

agramfort, 2021-03-28:
let me try to resurrect things with @l-omar-chehab

what I would do now is adjust our pipeline to produce "ICA features from
bids files : topos, PSDs etc."
from BIDS formatted data.

cause the way you extract the features has a lot of importance.

Give me a few days

>

hoechenberger, 2021-03-28:
@agramfort Happy to talk about this tomorrow 😊

l-omar-chehab, 2021-03-29:
Likewise!

adam2392, 2021-03-31:
Sounds good! I will ping some folks on my end to see if they can help since I'm stretched thin atm.

agramfort, 2021-03-31:
sent you and invite @adam2392

adam2392, 2021-04-11:
@jacobf18 an undergrad at JHU is interested in getting this going. I can help oversee and also make contributions.

He found the following issue: https://github.com/sccn/ICLabel/issues/14 where there was some discussion and also links to downstream repos: https://github.com/lucapton/ICLabel-Train and https://github.com/lucapton/ICLabel-Dataset. It seems that at least we have reference code and documentation of how EEGLab did this. 

### Plan Moving Forward

Ideally we can match EEGLab's feature extractions as much as possible, since they've already largely validated that pipeline. If we are missing any 1 feature, or have some technical differences due to how EEGLab/MNE backend, then we should note that.

@jacobf18, we should follow @agramfort 's suggestions linked here: https://github.com/mne-tools/mne-python/issues/9206#issuecomment-808926573

0. @agramfort can you invite @jacobf18 to that private repo?
1. a good first goal is to identify how we can extract data from the database
2. Next, if the data from database is not in BIDS format, we should have a pipeline to convert to BIDS format using `mne-bids`.
3. Then we take data from BIDS format and try to mirror the feature extraction that EEGlab is doing
4. Finally, once we have this process robust and documented, we evaluate models, such as CNN, NN-variants and RF using a train, validation, test set K-Fold cross validation. 
5. Save weights and produce figures to convince us what is best model.

Lmk if this sounds right?

agramfort, 2021-04-11:
sounds like a plan !

I invited @jacobf18 <https://github.com/jacobf18> to the private repo

>

larsoner, 2021-04-12:
Sounds like there is nothing to do at the MNE end for now so I'll close, but if I'm mistaken I'm happy to reopen

closed 2021-04-12 by larsoner

yjmantilla, 2021-05-03:
I dont know if the project will use the biharmonic spline interpolation used for the Scalp topography images of ICLABEL but if it is of any help I got some code for that at https://github.com/yjmantilla/gdatav4 .

yjmantilla, 2021-05-29:
Well I managed to produce some of the features as eeglab in [this repo](https://github.com/yjmantilla/iclabelpy) but still having trouble. For more details see https://github.com/sccn/ICLabel/issues/14

adam2392, 2021-11-04:
@Jacobf18 has done some great work demonstrating he can reproduce the EEGLab feature extraction methods. We are in the process of now reproducing their CNN model pipeline. Once that is done, we intend on:

1. exploring additional ways of improving the model, by either simplifying it to something like a RF, or adding modern architecture complexities
2. additionally validating on some data @agramfort was kind enough to provide for us.

Once that is done, I think we intend on re-exploring adding this feature into MNE-Python.

ctrltz, 2021-11-05:
Thank you for the reply. I think that this feature could be a really useful tool to include in automatic preprocessing pipelines, and it is very cool that you were able to get some progress on it. If there is any way for me to contribute to this porting process, I would be really glad to participate. Otherwise just hoping to see ICLabel included in MNE someday, and good luck with the project!