mne-tools/mne-python #9209: Add Raw, Epochs, and Evoked method to fire up MNE Coreg GUI

hoechenberger, 2021-03-28:
I'd love to do one of:

```python
epochs.mri_coreg(...)
epochs.coreg_mri(...)
epochs.coregistration(...)
```
to bring up `mne.gui.coregistration()` with some of its parameters automatically filled in based on the respective `Raw/Epochs/Evoked` instance.

Strong feelings against this?

larsoner, 2021-03-29:
Currently we have `mne.gui.coregistration`. I guess its API is limited by having to pass filenames rather than instances themselves, but that's easy enough to change. I'd rather just update this function.

agramfort, 2021-03-29:
>
> Currently we have mne.gui.coregistration. I guess its API is limited by
> having to pass filenames rather than instances themselves, but that's easy
> enough to change. I'd rather just update this function.
>

+1

closed 2022-02-21 by larsoner