mne-tools/mne-python #9219: Stable docs out of sync with maint branch?

hoechenberger, 2021-03-30:
I just realized that the [0.22.1 changelog](https://mne.tools/stable/whats_new.html#version-0-22-1) on our website is out of sync with the respective RST file, see e.g. [this section](https://github.com/mne-tools/mne-python/blob/1987cea8e28d2a42224f55adb7d6e175bf29f8b6/doc/changes/0.22.inc#L15-L17)

Is that a bug? Or am I missing something?

larsoner, 2021-03-30:
That probably should have been added to a new 0.22.2 section

larsoner, 2021-03-30:
... oh wait we haven't released 0.22.1 yet. Weird that it's not up to date, maybe there was some problem with the CircleCI build

larsoner, 2021-03-30:
I'm pushing hotfixes, current CircleCI build is at least running

https://app.circleci.com/pipelines/github/mne-tools/mne-python/7567/workflows/99cda811-20c3-4fab-a5a2-57428d19fd14/jobs/26639

I'll close once it's updated

larsoner, 2021-03-31:
All better

closed 2021-03-31 by larsoner