mne-tools/mne-python #9224: Reading OpenBCI EEG data with MNE python

SreeHaran, 2021-03-30:
#### Describe the new feature or enhancement
OpenBCI cyton board is one of the popular tool used for collecting EEG data from brain. The EEG data collected from the cyton board are stored in txt format with the help of OpenBCI GUI. Currently, MNE doesn't have any dedicated reader for function for OpenBCI data. It would be helpful if there's a separate class containing methods for reading, visualizing the EEG data from OpenBCI cyton board.

#### Additional comments
You can find Sample dataset for Cyton Board - 8 channels(without diasy module) [here](https://github.com/SreeHaran/BCI/tree/main/Collected%20Signals)

cbrnr, 2021-03-30:
We've already had such a request a while back in #3134. Have you checked out the package mentioned there? It might be outdated and unmaintained, but it's probably worth checking it out first.

SreeHaran, 2021-03-30:
Hey, @cbrnr in [#3134](https://github.com/mne-tools/mne-python/issues/3134), they discussed about implementing this feature with OpenBCI_Python, but it has been deprecated by OpenBCI. Is there any plan for implementing this feature in other ways?

SreeHaran, 2021-03-31:
Ok thank you, I'll take a look at it...

drammock, 2021-04-02:
I've poked around online a bit, and have confirmed that the structure of the text files is not very consistent.  Here is one from an OpenBCI Ganglion that is very different from the sample files posted above by @SreeHaran

https://github.com/J77M/eeg-datasets/blob/master/eyes-one-channel/01.csv

In the OpenBCI docs, they also imply, in [their EEGLAB tutorial](https://openbci.com/community/eeglab-tutorial-import-data/), that the number of rows to skip at the start of the file varies with the version of their software GUI that was used to record the data.  For EEGLAB they recommend using MATLAB's `csvread` function.  So it's unlikely that we'll be able to do any better.  Closing as `WONTFIX`; users will need to use `pandas.read_csv()` or `numpy.loadtxt()` or similar, and handle units conversion / channel names / etc themselves, and then load the data into `RawArray`.

larsoner, 2021-04-02:
I'm inclined to reopen and add an entry to the EEG data format table, along the lines of what I recently recommended regarding XDF. We can at least increase discoverability and give people some hints in the right direction that way.