mne-tools/mne-python #9231: mne.preprocessing.interpolate_bads taking list of instances

agramfort, 2021-04-01:
the issue is the following:

we need to be able to concatenate raws or epochs with different sets of bads.

currently mne.concatenate_raws and mne.concatenate_epochs enforce (for good reasons)
that the list of bads are the same for all instances.

The question is how to simplify the life of users to it's easy to have such "concatenable" instances.

Currently some users interpolate all bads for all instances using interpolate_bads methods.
It works but I think it is not ideal. For me you need the big picture of all instances to make
the best decision. For example I think it's NOT a good idea to systematically interpolate a channel that
is bad for all runs, however interpolation certainly makes sense if one channel is bad for let's say 50%
of the runs.

Suggestion:

`insts = mne.preprocessing.interpolate_bads([inst1, inst2, ...], good_fraction=0.5)`

would just interpolate if a channel is good for at least 50% of the runs and otherwise the
channel would be marked as bad for all instances.

then all interpolate_bads method would use this new public function with `[self]` as parameter.

we'll need to be careful about inplace computation options.

thoughts?

drammock, 2021-04-01:
Some questions:

- Is good-fraction based on number of instances, or duration?
- is there a principled way to choose good_fraction?
- depending on how you answer the second question, this feels like it's getting close to autoreject-like functionality. Are we sure it belongs in MNE-Python?

agramfort, 2021-04-01:
>
> Some questions:
>
>    - Is good-fraction based on number of instances, or duration?
>
> it would be based on the number of instances passed. For example if you
pass 5 instances
if you have 3 instances where channel XXX is good then you keep it if
good_fraction < 0.6.

>
>    - is there a principled way to choose good_fraction?
>
>
well not really.... so far people use fraction = 0 or fraction = 1 it's not
very quantitative either.
what would you suggest to users?

>
>    - depending on how you answer the second question, this feels like
>    it's getting close to autoreject-like functionality. Are we sure it belongs
>    in MNE-Python?
>
> note that here we assume that bads have been marked. It's for example our
case with BIDS datasets where
bad channels are listed

drammock, 2021-04-01:
> it would be based on the number of instances passed. For example if you
> pass 5 instances if you have 3 instances where channel XXX is good then you keep it if good_fraction < 0.6.

I would personally base it on duration. If I want to concatenate 3 instances of duration 6 mins, 1 min, 1 min, I would want to weight the 6min instance more highly when deciding whether to keep channels.

> is there a principled way to choose good_fraction?

I haven't thought about it very hard, but was wondering if there's a way to take rank into account somehow. Like first gather the union of all bad channels and for each one see what fraction of the final duration it is marked "bad", then mark bads in descending order of their bad fraction, up to whatever point that it starts affecting your rank too much.  In other words decide `good_fraction` automatically.

> > Are we sure it belongs in MNE-Python?
> note that here we assume that bads have been marked.

I agree now that it makes more sense here than in autoreject.

larsoner, 2021-04-01:
Agreed that fraction should be of time and not of instances. For a first iteration I would just allow `good_fraction` as a float, we can always add a `good_fraction='auto'` later if we can think of a good principled way to do it. Though maybe the signature would be:
```
mne.preprocessing.interpolate_mark_bads(...)
```
because it will either interpolate or mark (for all instances) so that at the end of the operation `info['bads']` matches for all instances, and contains the bads that were bad more than `good_fraction` of the time.

PR #9241 (2021-04-05): ENH : add new interpolate_mark_bads function
closes #9231
closed 2021-04-05 by drammock