mne-tools/mne-python #9234: BUG: apply_function arg/kwarg handling

drammock, 2021-04-01:
`raw/epochs/evoked.apply_function` all have an `*args, **kwargs` in their signatures. But they don't work the way I think they should / the docs are rather misleading. Preparation:

```py
import os
import mne
from scipy.ndimage import gaussian_filter

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file, verbose=False, preload=False)
raw.crop(tmax=5).load_data()
```

`gaussian_filter1d` requires a `sigma` parameter in addition to the data. This works as expected:
```py
raw.apply_function(gaussian_filter1d, sigma=4)
```
This also works:
```py
kwargs = dict(sigma=4)
raw.apply_function(gaussian_filter1d, **kwargs)
```

This next one is expected to fail **if you already know how argument unpacking works**, and does fail (`4.` is passed as `picks` param of `apply_function`, so you get `picks must be a list of int or list of str, got a data type of float64`):
```py
args = (4.,)
raw.apply_function(gaussian_filter1d, *args)
```
But note that if `args = (4,)` (int instead of float) you get a different error, that `gaussian_filter1d` is missing required argument `sigma`).  From the perspective of a new user, this is really bad.  I see `*args` in the docstring, so I pass in `*args` and it fails, in 2 different ways depending on what seems like a trivial difference in input.

This also fails though might have been expected/tried by a new user (`gaussian_filter1d() got an unexpected keyword argument 'args'`)
```py
args = (4.,)
raw.apply_function(gaussian_filter1d, args=args)
```

This also fails due to python syntax restrictions, but might have been expected to work based on the docstring:
```py
raw.apply_function(gaussian_filter1d, verbose=None, *args)
```

So in sum we don't actually support passing `args` to the containing function.  Possible solutions:

1. change function signature from `*args, *kwargs` to `args, kwargs`, and accept only tuples for `args` and only dicts for `kwargs`
2. drop `*args` from the function signature, leave kwarg handling as it is

agramfort, 2021-04-01:
I would use functools.partial but I understand if you say it's too
complicated...

or maybe we document it clearly in the docstring?

>

drammock, 2021-04-01:
> I would use functools.partial but I understand if you say it's too complicated...

I would say the safest / easiest-to-maintain option is not to accept args/kwargs at all, and advise people to use `functools.partial`. This would require a deprecation cycle for sure; people's code would break.

I would say the most minimal change (not requiring deprecation, because it never worked) is to eliminate `*args` and keep `**kwargs`.

I lean toward the second option, it's less work for us now and also less work for users, and `**kwargs` already works so it's not that much burden to keep it around because it's not that complicated.

larsoner, 2021-04-01:
Agreed `**kwargs` is painless enough to keep supporting, and we should kill `*args` as a bugfix

agramfort, 2021-04-01:
Ok

hoechenberger, 2021-04-02:
+1 for keeping kwargs and dropping args

PR #9235 (2021-04-05): Fix apply function
closes #9234 

I've locally updated the test for `epochs.apply_function()` and I'm a bit surprised / wondering if there's another bug in there.  Marking as draft until I've figured it out one way or another.  It seems like when applied channel-wise it gets applied to an ndarray that is `(n_epochs, n_times)` and it's not clear to me that that's the right thing to do (maybe it should really be epoch-wise and channel-wise?)

cc @nordme and @vpeterson
closed 2021-04-05 by larsoner