mne-tools/mne-python #9242: Implement more baseline modes for epoch's apply_baseline

AaronAngJW, 2021-04-03:
Hi there,

Is it possible to implement more baseline mode methods for https://mne.tools/stable/generated/mne.Epochs.html#mne.Epochs.apply_baseline as with those available in https://mne.tools/stable/generated/mne.time_frequency.EpochsTFR.html#mne.time_frequency.EpochsTFR.apply_baseline ?

agramfort, 2021-04-03:
I am bit worried if we allow this unless we find a way to keep track of
what baseline method was used.
Currently epochs does not have a way to remember this.

>

AaronAngJW, 2021-04-03:
I see. So the next closest available option would be the 'combine' option under viz.plot_compare_evokeds? Which only works for more than 1 sensor for non-mean modes?

https://mne.tools/dev/generated/mne.viz.plot_compare_evokeds.html#mne.viz.plot_compare_evokeds

agramfort, 2021-04-04:
can you clarify what you would like to obtain? doing the same as
evoked.plot but with z-score baseline for example?

>

AaronAngJW, 2021-04-04:
Yes, z-score baseline. As I am also using z-score for TFR plotting, it would be nice to see the z-score equivalent activity for the epochs as well to get them as closely matched as possible.

agramfort, 2021-04-04:
the main reason why you want to use z-score on TFR is the 1/f power
spectral. If you don't do such things you pretty
much only see low frequencies. The situation  is very different for epochs
traces that don't suffer from this issue.