mne-tools/mne-python #9245: Install instructions should refer to environment.yml for the currently selected MNE version

hoechenberger, 2021-04-05:
Currently, the install instructions always guide you do download environment.yml from our main branch. However, ideally this would only be the case if you're viewing the development docs. For stable docs, we should link to the environment file located in the respective maintenance branch.

agramfort, 2021-04-05:
I agree. I recently had the issue with a new intern. He was installing the
pypi version with the
main environment file and it was broken (due to pyvista / pyvistaqt
changes).

>

larsoner, 2021-04-05:
> For stable docs, we should link to the environment file located in the respective maintenance branch.

Our `environment.yml` is meant to be backward compatible and reflect a dependency structure that works in the present day, so I don't see what problem this will solve.

> He was installing the pypi version with the main environment file and it was broken (due to pyvista / pyvistaqt changes).

I do not see how this would have fixed the problem above. This is due to PyVista making incompatible changes and/or us using their private attributes. So to fix this problem, we would have to make our `environment.yml` *frozen*, i.e., specify every version of every library for example using `pip freeze`.

This is the current `diff` between 0.22 and `main`:
```
--- a/environment_22.yml
+++ b/environment_main.yml
@@ -18,9 +18,6 @@ dependencies:
 - joblib
 - psutil
 - numexpr
-- traits
-- pyface
-- traitsui
 - imageio
 - tqdm
 - spyder-kernels>=1.10.0
@@ -34,6 +31,6 @@ dependencies:
 - nibabel
 - nilearn
 - python-picard
-- pyqt
+- pyqt!=5.15.3
```
In this instance, the trains/pyface/traitsui is inconsequential / a cleanup (mayavi will pull them in) so can be ignored. But the `pyqt!=5.15.3` one would have caused people problems, so we would have had to backport every `environment.yml` fix that we make now, too. So to me it creates more maintenance burden for us without any appreciable gain for our users.

drammock, 2021-04-05:
I agree with @larsoner; `environment.yml` installs stable version by default (you have to do extra stuff afterward to get dev version).  If it's not working we should correct it, but pointing people to old versions of the env file probably won't help and might hurt (e.g., if we've fixed dependency problems that arose since the last release).

agramfort, 2021-04-05:
I am fine to treat this as "won't fix" and it's clearly not easy...

>

hoechenberger, 2021-04-05:
Alright then, thanks for the explanation, @drammock and @larsoner!

closed 2021-04-05 by hoechenberger