mne-tools/mne-python #9249: new repr_html makes tutorials too long

drammock, 2021-04-05:
The new `_repr_html` for Info, Raw, Epochs, and Evoked is pretty cool. But it has a drawback, which is that *every* time an instance is returned at the end of a tutorial cell, we get the full HTML repr (13 lines for a Raw).  In some tutorials this happens multiple times, the extra information isn't always helpful, and it breaks up the flow of the lesson.  For example, in [this tutorial](https://mne.tools/dev/auto_tutorials/intro/plot_20_events_from_raw.html) the HTML repr of Raw happens twice, once after a call of `raw.crop(tmax=60).load_data()` and once after `raw.set_annotations()`.  The first occurence is maybe arguably useful (though not directly relevant to the tutorial's purpose) but IMO the second one is clearly not helpful.  I scanned through the first ~15-20 tutorials and found that about 1/3 of them have 2 instances of a Raw HTML repr; I didn't check closely but I'd guess that many of them are extraneous / unneeded.

Possible solutions:

1. live with it
2. go around putting `_ = ` before all the cases where we want to suppress the HTML output
3. is there a way to only show the nice table when the object is explicitly `print`ed?

WDYT @vagechirkov @hoechenberger et al?

larsoner, 2021-04-05:
I would suggest a different solution:

- Enhance sphinx-gallery to allow per-block `capture_repr` config

In principle this should only require a few lines in Sphinx-gallery. And our default config can be allow-list style by setting `capture_repr=()` in `conf.py`, and in the blocks where we want it, enable it via `# sphinx_gallery_capture_repr = ('_repr_html_',)` or so (or vice-versa conf/inline for blocklist-style).

hoechenberger, 2021-04-05:
We could display a short summary and collapse the table by default.

Here's an example on a collapsible HTML representation using ipywidgets:
https://matthewrocklin.com/blog/2019/07/04/html-repr#logs

vagechirkov, 2021-04-06:
I agree with  @hoechenberger. I think it is better to collapse by default the long representations. I will create the draft PR with this feature

drammock, 2021-04-06:
I keep going back and forth in my head whether collapsing by default is a good approach or not.  In some contexts (like in Reports) I think we want *expanded* to be the default?  In others (many, but not all, occurrences in the tutorials) we either don't want/need to see the repr, or we at least want it collapsed by default.

So assuming my option 3 isn't actually possible, I sort of lean toward what @larsoner suggests, even though it compels us to add extra sphinx-gallery control comments to the tutorials... assuming that (1) those comments will get filtered out of the rendered HTML tutorial like the `sphinx_gallery_thumbnail_number` comments do, and (2) we can do it cell-by-cell rather than document-by-document.  If the comment filtering doesn't happen, we might as well just use `_ = whatever` since that's less to type and (possibly?) no more or less confusing to the reader.

larsoner, 2021-04-06:
Comment filtering should come for free but if it doesn't it should be at most one more SG line to update

larsoner, 2021-04-06:
... but looking at the existing options there are no block specific comments in SG yet, but I don't see why we can't have/add them for this use case

PR #12064 (2023-10-04): Collapsible html repr for raw/info
closes #9255 (supersedes)
closes #9249

@vagechirkov I marked you as co-author on the commit since I basically re-implemented what you did in #9255 (changes on main were too drastic to do a simple rebase)