mne-tools/mne-python #9258: `apply_function` method for Epochs object

hubertjbanville, 2021-04-06:
In the braindecode project we've tried to unify the preprocessing of raw and epoched data under a single class (https://github.com/braindecode/braindecode/blob/master/braindecode/datautil/preprocess.py#L61). We currently use a hybrid approach that relies on `Raw.apply_function()` when possible, and would benefit from having an equivalent method for the `Epochs` object (discussion here: https://github.com/braindecode/braindecode/issues/160). 

If of interest I'd be happy to submit a PR for this - I would just like to know whether it would make sense to share the implementation for `Raw` and `Epochs` via something like a utility function or a mixin?

Thanks!

drammock, 2021-04-06:
`mne.Epochs.apply_function()` was added last month in #9088, and was patched yesterday in #9235.  Can you try current MNE-Python main to see if it works for your use case?

hubertjb, 2021-04-06:
Oops I missed that, sorry. It does work as expected for me on the latest main, thanks!

closed 2021-04-06 by drammock