mne-tools/mne-python #9260: RFC: Epochs.equalize_event_counts() should operate on all events by default

hoechenberger, 2021-04-07:
Currently, `Epochs.equalize_event_counts()` has a positional argument `event_ids`, which expects a list of event names (for some reason called "event ids" in the docstring) whose counts to equalize.

Now a common scenario is that I have, say, 2 or 3 conditions (event types), and before calculating their contrast, I want to equalize their count. Currently, I'll have to do something like this:

```python
epochs.equalize_event_counts(epochs.event_id.keys())
```

I'd like to be able to just do this instead
```python
epochs.equalize_event_counts()
```

To achieve this, the signature of the method should be changed from currently:
```python
    def equalize_event_counts(self, event_ids, method='mintime'):
```

to
```python
    def equalize_event_counts(self, event_ids=None, method='mintime'):
```

and if `event_ids is None`, we simply set `event_ids = list(self.event_id.keys())`

WDYT?

agramfort, 2021-04-07:
Sounds good 👌

larsoner, 2021-04-07:
Sounds good to me

PR #9261 (2021-04-07): MRG, ENH: Epochs.equalize_event_counts() operates on all events by default
Fixes #9260
closed 2021-04-07 by drammock