mne-tools/mne-python #9263: MNE Python stc.plot(),  draws the brain as translucent

Mirrabella, 2021-04-07:
When I use stc.plot(), this function draws the brain as translucent, and it is seems very stange and the picture is poorly perceived.

Can you please tell me how to change the transparency parameter?

MNE version 23.0

I tried to used View — Toggle Eye Dome Lighting, but it did not help,  picture still seems strange too. :(
[image]
[image]

GuillaumeFavelier, 2021-04-07:
Hello @Mirrabella and thank you for reporting your issue. I would like to know more about your configuration, it will help me investigate the origin of this behaviour.

If you don't mind, could you share the output of:

```sh
$ python -c "import mne; mne.sys_info()"
```

and:

```sh
$ python -c "import pyvista; print(pyvista.Report())"
```

It would be helpful if you could also share a minimal code snippet giving this output so that I can try to reproduce on my end.

GuillaumeFavelier, 2021-04-07:
From what I can tell, you seem to use the `pyvista` 3d backend with the old version of the GUI of `stc.plot()` since those sliders have been migrated from `vtk` to `PyQt5` in https://github.com/mne-tools/mne-python/pull/8862/.

Mirrabella, 2021-04-08:
Hello and thank you very much for your answer.

My configuration on printscreen.

Code is from example MNE Python:

import os.path as op

import os

import mne

from mne.datasets import sample

data_path = sample.data_path()

# the raw file containing the channel location + types

raw_fname = data_path + '/MEG/sample/sample_audvis_raw.fif'

# The paths to Freesurfer reconstructions

subjects_dir = data_path + '/subjects'

os.environ['SUBJECTS_DIR'] = subjects_dir

subject = 'sample'

raw = mne.io.read_raw_fif(raw_fname)

events = mne.find_events(raw = raw, stim_channel = 'STI 014')

event_id = dict(aud_r=2, vis_r=4)

epochs = mne.Epochs(raw=raw, events = events, event_id = event_id)

evoked = epochs.average()

conductivity = [0.3] # for single layer

model = mne.make_bem_model(subject=subject, ico=4, conductivity=
conductivity, subjects_dir=subjects_dir, verbose=None)

bem = mne.make_bem_solution(model)

trans = data_path + '/MEG/sample/sample_audvis_raw-trans.fif'

src = mne.setup_source_space(subject, spacing='oct4', add_dist='patch',
subjects_dir=subjects_dir)

fwd = mne.make_forward_solution(raw_fname, trans=trans, src=src,
bem=bem, meg=True,
eeg=False, mindist=5.0, n_jobs=1, verbose=True)

cov = mne.compute_covariance(epochs, method='auto')

inv = mne.minimum_norm.make_inverse_operator(raw.info, fwd, cov, loose=0.2)

stc = mne.minimum_norm.apply_inverse(evoked, inv, lambda2=1. / 9.)

stc.plot(hemi='both')

Thank you in advance:)

Mirrabella, 2021-04-08:
> Hello @Mirrabella and thank you for reporting your issue. I would like to know more about your configuration, it will help me investigate the origin of this behaviour.
> 
> If you don't mind, could you share the output of:
> 
> ```shell
> $ python -c "import mne; mne.sys_info()"
> ```
> 
> and:
> 
> ```shell
> $ python -c "import pyvista; print(pyvista.Report())"
> ```
> 
> It would be helpful if you could also share a minimal code snippet giving this output so that I can try to reproduce on my end.

Hello and thank you very much for your answer.

My configuration on printscreen.

[image]
[image]

Code is from example MNE Python:

import os.path as op
import os
import mne
from mne.datasets import sample

data_path = sample.data_path()
# the raw file containing the channel location + types
raw_fname = data_path + '/MEG/sample/sample_audvis_raw.fif'

# The paths to Freesurfer reconstructions
subjects_dir = data_path + '/subjects'
os.environ['SUBJECTS_DIR'] = subjects_dir
subject = 'sample'
raw = mne.io.read_raw_fif(raw_fname)
events = mne.find_events(raw = raw, stim_channel = 'STI 014')
event_id = dict(aud_r=2, vis_r=4)
epochs = mne.Epochs(raw=raw, events = events, event_id = event_id)
evoked = epochs.average()
conductivity = [0.3] # for single layer
model = mne.make_bem_model(subject=subject, ico=4, conductivity= conductivity, subjects_dir=subjects_dir, verbose=None)
bem = mne.make_bem_solution(model)
trans = data_path + '/MEG/sample/sample_audvis_raw-trans.fif'
src = mne.setup_source_space(subject, spacing='oct4', add_dist='patch', subjects_dir=subjects_dir)
fwd = mne.make_forward_solution(raw_fname, trans=trans, src=src, bem=bem, meg=True, eeg=False, mindist=5.0, n_jobs=1, verbose=True)
cov = mne.compute_covariance(epochs, method='auto')
inv = mne.minimum_norm.make_inverse_operator(raw.info, fwd, cov, loose=0.2)
stc = mne.minimum_norm.apply_inverse(evoked, inv, lambda2=1. / 9.)
stc.plot(hemi='both')

Thank you in advance:)

GuillaumeFavelier, 2021-04-08:
Unfortunately, I can't reproduce this transparency bug on my machine right now but I can still guess that it could come from `vtk` or `mesa` :thinking: 

I'm less familiar with `mesa` but I can try to help for `vtk 9.0.1`:

1) You could use the latest version of MNE [conda configuration file](https://github.com/mne-tools/mne-python/blob/93359566261be5c3fe7ceef8cb1fefce910cca52/environment.yml) to create a new environment. It uses at least `python 3.8` so I hope you don't have any issue with that (it's required since `mne 0.21.0` anyway) but I expect it to fetch the latest `vtk` successfully. The installation procedure for each system is described on [MNE's website](https://mne.tools/stable/install/mne_python.html#id5):

<details>
<summary><b>environment.yml</b></summary>

```
name: mne
channels:
- conda-forge
dependencies:
- python>=3.8
- pip
- numpy
- scipy
- matplotlib
- numba
- pandas
- xlrd
- scikit-learn
- h5py
- pillow
- statsmodels
- jupyter
- joblib
- psutil
- numexpr
- traits
- pyface
- traitsui
- imageio
- tqdm
- spyder-kernels>=1.10.0
- imageio-ffmpeg>=0.4.1
- vtk>=9.0.1
- pyvista>=0.24
- pyvistaqt>=0.2.0
- mayavi
- PySurfer
- dipy
- nibabel
- nilearn
- python-picard
- pyqt
- mne
- mffpy>=0.5.7
```
</details>

2) If you know how to manage an environment with both `conda` and `pip` or if 1) does not work, you could try:

```
conda uninstall vtk
pip install --upgrade vtk
```

This will effectively fetch the latest version of `vtk` available on PyPI but I have to warn you that it is not a stable solution (as package coherence is not fully managed by conda).

Mirrabella, 2021-04-09:
> Unfortunately, I can't reproduce this transparency bug on my machine right now but I can still guess that it could come from `vtk` or `mesa` 
> 
> I'm less familiar with `mesa` but I can try to help for `vtk 9.0.1`:
> 
> 1. You could use the latest version of MNE [conda configuration file](https://github.com/mne-tools/mne-python/blob/93359566261be5c3fe7ceef8cb1fefce910cca52/environment.yml) to create a new environment. It uses at least `python 3.8` so I hope you don't have any issue with that (it's required since `mne 0.21.0` anyway) but I expect it to fetch the latest `vtk` successfully. The installation procedure for each system is described on [MNE's website](https://mne.tools/stable/install/mne_python.html#id5):
> 
> environment.yml
> 1. If you know how to manage an environment with both `conda` and `pip` or if 1) does not work, you could try:
> 
> ```
> conda uninstall vtk
> pip install --upgrade vtk
> ```
> 
> This will effectively fetch the latest version of `vtk` available on PyPI but I have to warn you that it is not a stable solution (as package coherence is not fully managed by conda).

I tried to make the settings according to point 1. But now an error occurs. I would be very grateful if you could give me some more advice on how to fix it.

As I understand it, there are some problems with mesa (I'm not very strong in this :()

[image]

GuillaumeFavelier, 2021-04-09:
Hm... It seems that either mesa cannot be setup properly or no window can be opened, are you by any chance using a headless server? If that's the case you might be interested in:

https://mne.tools/stable/install/mne_python.html?highlight=server#installing-to-a-headless-server

closed 2022-02-21 by larsoner