mne-tools/mne-python #9265: SNIRF loader not working with the official samples

HanBnrd, 2021-04-08:
#### Describe the bug
The function [*read_raw_snirf*](https://mne.tools/dev/generated/mne.io.read_raw_snirf.html#mne.io.read_raw_snirf) returns errors with the official samples from https://github.com/fNIRS/snirf-samples/.

#### Steps to reproduce
With [*Simple_Probe.snirf*](https://github.com/fNIRS/snirf-samples/blob/master/basic/Simple_Probe.snirf):
```python
from mne.io import read_raw_snirf

snirf_intensity = read_raw_snirf('Simple_Probe.snirf')
```

With [*minimum_example.snirf*](https://github.com/fNIRS/snirf-samples/blob/master/test/minimum_example.snirf):
```python
from mne.io import read_raw_snirf

snirf_intensity = read_raw_snirf('minimum_example.snirf')
```

#### Expected results
```
Loading Simple_Probe.snirf
```
and
```
Loading minimum_example.snirf
```
respectively.

#### Actual results
```
Loading Simple_Probe.snirf
...\snirf.py:3: RuntimeWarning: Non uniform sampled data not supported.
  snirf_intensity = read_raw_snirf('Simple_Probe.snirf')
...\snirf.py:3: RuntimeWarning: Unable to extract sample rate from SNIRF file.
  snirf_intensity = read_raw_snirf('Simple_Probe.snirf')
Traceback (most recent call last):
  File "...\snirf.py", line 3, in <module>
    snirf_intensity = read_raw_snirf('Simple_Probe.snirf')
  File "...\mne-python\mne\io\snirf\_snirf.py", line 43, in read_raw_snirf
    return RawSNIRF(fname, preload, verbose)
  File "<decorator-gen-248>", line 24, in __init__
  File "...\mne-python\mne\io\snirf\_snirf.py", line 112, in __init__
    sources = [s.decode('UTF-8') for s in sources]
  File "...\mne-python\mne\io\snirf\_snirf.py", line 112, in <listcomp>
    sources = [s.decode('UTF-8') for s in sources]
AttributeError: 'numpy.ndarray' object has no attribute 'decode'
```
and
```
Loading minimum_example.snirf
...\snirf.py:3: RuntimeWarning: Non uniform sampled data not supported.
  snirf_intensity = read_raw_snirf('minimum_example.snirf')
...\snirf.py:3: RuntimeWarning: Unable to extract sample rate from SNIRF file.
  snirf_intensity = read_raw_snirf('minimum_example.snirf')
Traceback (most recent call last):
  File "...\snirf.py", line 3, in <module>
    snirf_intensity = read_raw_snirf('minimum_example.snirf')
  File "...\mne-python\mne\io\snirf\_snirf.py", line 43, in read_raw_snirf
    return RawSNIRF(fname, preload, verbose)
  File "<decorator-gen-248>", line 24, in __init__
  File "...\mne-python\mne\io\snirf\_snirf.py", line 112, in __init__
    sources = [s.decode('UTF-8') for s in sources]
TypeError: iteration over a 0-d array
```
respectively.

#### Additional information
This [tutorial from MNE-NIRS](https://mne.tools/mne-nirs/auto_examples/plot_90_snirf.html) works fine on my machine.

HanBnrd, 2021-04-08:
The SNIRF loader is also not working if the [optional](https://github.com/fNIRS/snirf/issues/41#issuecomment-620159229) *sourcePos3D* field does not exist in the SNIRF file.

```
Loading nback.snirf
Traceback (most recent call last):
  File "...\snirf.py", line 3, in <module>
    snirf_intensity = read_raw_snirf('nback.snirf')
  File "...\mne-python\mne\io\snirf\_snirf.py", line 43, in read_raw_snirf
    return RawSNIRF(fname, preload, verbose)
  File "<decorator-gen-248>", line 24, in __init__
  File "...\mne-python\mne\io\snirf\_snirf.py", line 119, in __init__
    assert len(sources) == srcPos3D.shape[0]
IndexError: tuple index out of range
```

HanBnrd, 2021-04-16:
Hi, just pinging @rob-luke.
I will try to have a look at it ASAP.

HanBnrd, 2021-04-16:
I created a separate issue for the sourcePos3D problem: #9308

rob-luke, 2021-04-17:
We should ensure we follow the SNIRF standard. However, from memory not all files in the examples repo do. What specific part of our reader is diverging from the standard?

HanBnrd, 2021-04-17:
Thanks @rob-luke, you are right, my bad. Both files do not pass the [*snirf_validator.py*](https://github.com/BUNPC/python_snirf_validator/blob/master/snirf_validator.py).

rob-luke, 2021-04-17:
That's good to know. Thanks for checking this. I probably wouldn't trust that validator either. Just the spec is valid

closed 2021-04-17 by HanBnrd

rob-luke, 2021-04-22:
I just received some SNIRF files from NIRx. So I will test and ensure these load in to our reader.