mne-tools/mne-python #9272: UnicodeDecodeError: 'charmap' codec can't decode byte 0x81 in position 6437: character maps to <undefined>

leticiabragas2, 2021-04-09:
Can anyone tell me HOW TO SOLVE? please!

sample_data_folder = mne.datasets.testing.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'NihonKohden','MB0400FU.EEG')
raw = mne.io.read_raw_nihon(sample_data_raw_file, verbose=False)

raw = mne.io.read_raw_nihon(sample_data_raw_file, verbose=False)
events_file = os.path.join(sample_data_folder, 'NihonKohden','MB0400FU.EEG')
events = mne.read_events(events_file)

  File "E:\Usuários\Letícia\Desktop\test\ArquivoAlfred.py", line 60, in <module>
    events = mne.read_events(events_file)

  File "<decorator-gen-157>", line 21, in read_events

  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\event.py", line 265, in read_events
    lines = np.loadtxt(filename, dtype=np.float64).astype(int)

  File "C:\ProgramData\Anaconda3\lib\site-packages\numpy\lib\npyio.py", line 1112, in loadtxt
    first_line = next(fh)

  File "C:\ProgramData\Anaconda3\lib\encodings\cp1252.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]

UnicodeDecodeError: 'charmap' codec can't decode byte 0x81 in position 6437: character maps to <undefined>

cbrnr, 2021-04-09:
This is probably related to #9267. Let's wait if @fraimondo or @larsoner can help.

cbrnr, 2021-04-09:
I can reproduce on `main`.

cbrnr, 2021-04-09:
Alright, the problem is that you are trying to read a file with `mne.read_events` which is not an events file. `MB0400FU.EEG` contains EEG data in Nihon Khoden format.

cbrnr, 2021-04-09:
Closing this issue, feel free to re-open or ask further usage questions in our forum.

closed 2021-04-09 by cbrnr