mne-tools/mne-python #9281: ENH : Building a test for checking the ylims in plot_evoked_topo

ramkpari, 2021-04-11:
As discussed in PR https://github.com/mne-tools/mne-python/pull/9207#issuecomment-815302885,  We need a test to check the ylims used for the individual subplots for `plot_evoked_topo`.  This would identify and prevent any further issues arising with ylims in the future from newer versions of python. 

Since the `plot_evoked_topo` has unified=true https://github.com/mne-tools/mne-python/pull/9207#issuecomment-814725032, It looks like it might not be possible to unpack the figure/axes object to access the individual ylims directly. 

One idea that I had proposed earlier to @jasmainak  was to build a `debug` parameter into `plot_evoked_topo` to be able to access all the individual parameters for the purpose of running tests before the creation of the figure object.  Thoughts @larsoner @agramfort @drammock?

agramfort, 2021-04-11:
also thought about this. testing unified=False is maybe easier has you
should get the axes with fig.axes

ramkpari, 2021-04-11:
Currently, there seems to be no call to `plot_evoked_topo` as a parameter. The `unified=true` param only comes into place when when `plot_topo` is called here https://github.com/mne-tools/mne-python/blob/53a79ec1b6d3c42951263a1e1aa6e7f9a4c2c6be/mne/viz/topo.py#L787.  Swapping that around to `True` and it seems like the current version of `plot_topo` isn't handling it well, with multiple exceptions for pos and data lines. Temporarily adding pos and data lines in at L165 with 
```python
ax.pos=pos[idx]
ax.data_lines=list()
``` 
seems to temporarily push the issue upto L481, where there's an exception regarding the lack of a unified axes object. 

Basically, the current state of `plot_topo` can't handle a unified = False call from `plot_evoked_topo`. Do you think I should work further along this route to test and work on implementing unified = False or look at other methods?

jasmainak, 2021-04-13:
so `plot_topo` is just a wrapper around `_iter_topography`. Does unified=False work with `iter_topography` or that fails as well?

ramkpari, 2021-04-15:
> so `plot_topo` is just a wrapper around `_iter_topography`. Does unified=False work with `iter_topography` or that fails as well?

The public-facing function  `iter_topography` doesn't allow `unified=False` but modifying it does allow `_iter_topography` to technically work with `unified=False`.