mne-tools/mne-python #9290: Rename files in tutorials?

cbrnr, 2021-04-13:
I have always been wondering why our tutorials are called `plot_*`, even though many tutorials do not produce a single plot. @hoechenberger pointed me to the reason, which is that this is just the default Sphinx naming scheme. However, we could change that (https://sphinx-gallery.github.io/stable/configuration.html#build-pattern). In fact, we only have `plot_*.py` files in our gallery folders, so we could easily create a rule that build all `*.py` files. WDYT?

hoechenberger, 2021-04-13:
+1 on renaming; even though I believe maybe we **should** keep some kind of prefix, like `tut_*`?

cbrnr, 2021-04-13:
If you want to prepare for a distant future case where we suddenly want to put a `.py` file that shouldn't get processed, I'd do it when we need it (YAGNI).

sappelhoff, 2021-04-13:
+1 on renaming -> as a newbie, I was confused by this naming convention - and now (being a bit more experienced), I think that it would look/read nicer to get rid of the prefix, if it's technically no problem.

>  even though I believe maybe we should keep some kind of prefix, like tut_*?

you mean for globally searching tutorial files?

Personally I would think that having them in the `tutorials/` directory would suffice :thinking:

hoechenberger, 2021-04-13:
> Personally I would think that having them in the `tutorials/` directory would suffice 🤔

ok, convinced :)

larsoner, 2021-04-13:
There is a downside to this renaming in that if people have bookmarks to the existing ones those links will break. But I agree the URL/filename readability benefit is probably worth it. WDYT @drammock ?

To do this the process can just be to update the sphinx-gallery conf pattern to be anything not prefixed by `_sgskip` (`'filename_pattern': '^((?!sgskip).)*$'`), rename the few examples that currently don't start with `plot_*` to `*_sgskip`, and rename all `plot_*` to no longer have that prefix. This would follow the matplotlib pattern

https://github.com/matplotlib/matplotlib/blob/master/doc/conf.py#L155

agramfort, 2021-04-13:
note that we have some examples that do not contain plot_ (I think about the RSA decoding example).

Also I foresee that google will need to reindex everything. So we'll have a lot of broken links. You can
find many of my slides online and you will break on the URL they contain

larsoner, 2021-04-13:
> note that we have some examples that do not contain plot_ (I think about the RSA decoding example).

These should be taken care of by my suggested `_sgskip` process above

> Also I foresee that google will need to reindex everything. So we'll have a lot of broken links. You can
find many of my slides online and you will break on the URL they contain

It just occurred to me that I could write ~30 lines in `conf.py` to handle this by writing out `plot_*` HTML redirects for all of our examples. I think this should make Google and your slides happy, and also make it so that in the future if we want to change page locations we can trivially (in one line) add arbitrary redirects from old-to-new names.

agramfort, 2021-04-13:
if you want to do this no objection.

>

larsoner, 2021-04-13:
Okay I'll see how bad it is!

drammock, 2021-04-13:
I love it when issues get resolved before I'm done with breakfast! FYI this is something I tried to do as part of the big doc restructuring in 2019. At that time my plan was to ditch `plot_` and have non-plotting tutorials end with `_`. But `_sgskip` seems fine too.

I should note that some, but not all, of the tutorials have had a number added to their filename, to enforce a specific sort order within their category. With the links changing anyway, now would be a good time to add numbers to the rest of them.

cbrnr, 2021-04-13:
I'd actually prefer the `_` suffix over `_sgskip` because it looks more Pythonic 😄 - but it's not that important. Adding numbers to all examples is a great idea.

The few examples that currently don't start with `plot_` - how are they rendered? What is the reason we don't "plot" them?

drammock, 2021-04-13:
The reason is long execution time. They are rendered by Sphinx from rst to html without running the py code or capturing output.

PR #9292 (2021-04-20): MAINT: Rename examples and tutorials
Adds automated redirect from `plot_*` to all of our examples, and renames them. @drammock for:

> I should note that some, but not all, of the tutorials have had a number added to their filename, to enforce a specific sort order within their category. With the links changing anyway, now would be a good time to add numbers to the rest of them.

I implemented (but did not test) some `custom_redirects` code that you can use. I think you'll want to adapt it to take care of the number renaming. I added the conditional `fname not in needed_plot_redirects.values() [... cut]
closed 2021-04-20 by larsoner