mne-tools/mne-python #9293: Incorrect number of samples error when loading a Synamps Neuroscan Curry file

arnevdk, 2021-04-13:
#### Describe the bug
When trying to read the data of a Neuroscan Synamps `.dat` file containing a recording with 128 EEG-electrodes, I am presented with the following error:
```python
RuntimeError: Incorrect number of samples (7623792 != 24999876), please report this error to MNE-Python developers
```
The file loads correctly in Matlab using EEGLAB and the loadcurry plugin.
I believe this issue to be different than #8391, since the number of channels looks correct and the reported number of samples is a lot smaller than the actual number of samples.

#### Steps to reproduce
```python
import mne
mne.io.read_raw_curry('subject_01.dat', preload=True)
```

I cannot legally share the `.dat` data file, but the accompanying  metadata files in the same directory can be found here:
*  `.ceo`: https://cloud.vandenkerchove.com/index.php/s/4L8dKCNtCHwwSLw
*  `.dap`: https://cloud.vandenkerchove.com/index.php/s/MKKxGZYkK4MYQET
*  `.rs3`: https://cloud.vandenkerchove.com/index.php/s/oYBzFA6RMDBAg4b

#### Expected results
The file loads correctly

#### Actual results
```
Leaving device<->head transform as None (no landmarks found)
Reading 0 ... 1748799  =      0.000 ...  1748.799 secs...
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<decorator-gen-242>", line 24, in read_raw_curry
  File "/data/Workspace/PhD/mne-python/mne/io/curry/curry.py", line 470, in read_raw_curry
    return RawCurry(fname, preload, verbose)
  File "<decorator-gen-243>", line 24, in __init__
  File "/data/Workspace/PhD/mne-python/mne/io/curry/curry.py", line 503, in __init__
    super(RawCurry, self).__init__(
  File "<decorator-gen-179>", line 24, in __init__
  File "/data/Workspace/PhD/mne-python/mne/io/base.py", line 254, in __init__
    self._preload_data(preload)
  File "/data/Workspace/PhD/mne-python/mne/io/base.py", line 534, in _preload_data
    self._data = self._read_segment(
  File "<decorator-gen-181>", line 24, in _read_segment
  File "/data/Workspace/PhD/mne-python/mne/io/base.py", line 419, in _read_segment
    _ReadSegmentFileProtector(self)._read_segment_file(
  File "/data/Workspace/PhD/mne-python/mne/io/base.py", line 2013, in _read_segment_file
    return self.__raw.__class__._read_segment_file(
  File "/data/Workspace/PhD/mne-python/mne/io/curry/curry.py", line 533, in _read_segment_file
    _read_segments_file(
  File "/data/Workspace/PhD/mne-python/mne/io/utils.py", line 222, in _read_segments_file
    raise RuntimeError('Incorrect number of samples (%s != %s), '
RuntimeError: Incorrect number of samples (7623792 != 24999876), please report this error to MNE-Python developers
```

#### Additional information
```
Platform:      Linux-5.11.11-200.fc33.x86_64-x86_64-with-glibc2.32
Python:        3.9.2 (default, Feb 20 2021, 00:00:00)  [GCC 10.2.1 20201125 (Red Hat 10.2.1-9)]
Executable:    /data/Workspace/PhD/mne-python/.venv/bin/python
CPU:           x86_64: 12 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.23.dev0
numpy:         1.20.2 {blas=openblas, lapack=openblas}
scipy:         1.6.2
matplotlib:    Not found

sklearn:       Not found
numba:         Not found
nibabel:       Not found
nilearn:       Not found
dipy:          Not found
cupy:          Not found
pandas:        Not found
mayavi:        Not found
pyvista:       Not found
vtk:           Not found
``

agramfort, 2021-04-14:
@DiGyt do you have time to have a look? thx

DiGyt, 2021-06-10:
Sorry for only noticing this now. I see what I can do...

arnevdk, 2021-06-16:
The file links have expired. These should stay active:

.ceo: https://cloud.vandenkerchove.com/index.php/s/sjo5axorkRXjAzq
.dap: https://cloud.vandenkerchove.com/index.php/s/ncKgzCc9NWcw4Bq
.rs3: https://cloud.vandenkerchove.com/index.php/s/kTjW4Z7daZ3y95n

arnevdk, 2021-06-16:
Other recordings from the same equipment do load correctly, so it still might be a data corruption issue. However, since the loaded data looks and functions fine in MNE as well as EEGLAB, could this be replaced by a warning?

agramfort, 2021-06-16:
@DiGyt can you guide us here?

DiGyt, 2021-06-22:
Just looked into it.
First of all, thanks @arnevdk for reporting this issue!

The problem occurs when loading the data from the `.dat` file with the "general" `mne.io` utility function `_read_segments_file` and I suspect it might be caused from the data type passed from curry's `read_segment_file` to said utility function. However, the data format specifications listed in the `.dap` file do not indicate any difference between this Curry dataset and other Curry datasets which don't produce this bug.

Unfortunately, this probably means that I wold require the `.dat` file, or at least a file of the same size and the same encoding in order to fix the issue. Would it be possible for you @arnevdk  to crop the `.dat` file or e.g. overwrite it with random numbers such that it still produces the error but you are allowed to share it with us? That would be great!

If the file loads correctly when ignoring the error, it would of course be a solution to remove the error or reduce it to a warning. But are you sure that when ignoring it, all the data is loaded correctly and there are e.g. no samples (or channels) missing at the end of the file? Of course I would rather make sure this is properly fixed that just omitting the error...

arnevdk, 2021-06-28:
The `.dat` file contains 830491200 bytes with encoding binary. Here is a version of the file with all bytes overwritten with 0: https://cloud.vandenkerchove.com/index.php/s/XaPobpicDkGQrJf . After inispection, it looks like  ignoring the error does not result in missing samples or channels, or corrupted samples or channels, but I might be missing some minor corruption. The number of samples loaded corresponds to the number of samples in the `.dap` file.

DiGyt, 2021-06-29:
Thank you very much for that @arnevdk . This will hopefully allow us to fix this issue soon.