mne-tools/mne-python #9295: BUG: next epochs when data not preloaded throws error

kingjr, 2021-04-13:
Issue only on dev

```python
import mne
from mne import io
from mne.datasets import sample
data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
event_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif'

raw = io.read_raw_fif(raw_fname)
events = mne.read_events(event_fname)
epochs = mne.Epochs(raw, events, preload=False)
next(epochs)
```

kingjr, 2021-04-14:
It is caused by https://github.com/mne-tools/mne-python/commit/59429c3b8022f61cf3539dc421cadff4ab675d24#diff-a8ad12a27877498bd9fe2ce1990d63852c08edca624268ae8d73dbac17b50b59

It can be fixed by simply using: 
```python
next(iter(epochs))
```

but it's a change of API, not sure you what you'd prefer (change of API versus change of current code)

larsoner, 2021-04-14:
Glad there is a workaround, I should be able to look into a proper fix later this week because I agree this is a bug

PR #9296 (2021-04-14): BUG: Fix bug with next without iter
Closes #9295

I'm not 100% sure we should support just doing `next` like this since:
```
>>> x = [1, 2]
>>> next(x)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'list' object is not an iterator
>>> next(iter(x))
1
```
But I can live with the simple fix in this PR, and we used to support it, so why not if it saves some people some pain.
closed 2021-04-14 by agramfort