mne-tools/mne-python #9297: find_events() returns times with offset

skjerns, 2021-04-14:
#### Describe the bug
I have a FIF file with a `STI101` trigger channel to which I apply `mne.find_events()` to extract the event markers. It seems to be that there is a mismatch of 5100 samples between the extracted events and the positions of the events when looking at the STI-Channel manually. Am I missing something? Why is there a delay to the raw STIM channel positions?

Do you have any idea why this is happening? 

sample file:
[events_raw.zip](https://github.com/mne-tools/mne-python/files/6312237/events_raw.zip)

```
import numpy as np
import mne

raw = mne.io.read_raw('events_raw.fif', preload=True)
# automatic extraction
events = mne.find_events(raw,stim_channel='STI101', shortest_event=1, consecutive=False)
# "manual" exctraction of events
manual_events = np.nonzero(raw.get_data('STI101'))[-1]
manual_events = manual_events[np.diff(np.pad(manual_events, [1, 0]))>1]

# consistent lag of 5100 samples
print(events[:,0]-manual_events)
```

however, I cannot reproduce this when I create the the same structure with an `RawArray`, which leads me to conclude that this is something inside of my file that I just can't figure out, but would badly need to fix.

```
raw = mne.io.read_raw('events_raw.fif', preload=True)
data = raw.get_data()
ch_types = ['grad', 'stim', 'syst']
ch_names = ['MEG2633', 'STI101', 'SYS201']
info = mne.create_info(ch_names=ch_names, sfreq=1000.0, ch_types=ch_types)
raw = mne.io.RawArray(data, info)
events = mne.find_events(raw,stim_channel='STI101', shortest_event=1, consecutive=False)

# now it works perfectly fine as expected, 0 lag
print(events[:,0]-manual_events)

```

#### Additional information
```
Platform:      Windows-10-10.0.19041-SP0
Python:        3.8.8 (default, Feb 24 2021, 15:54:32) [MSC v.1928 64 bit (AMD64)]
Executable:    C:\Users\skjerns\anaconda3\python.exe
CPU:           AMD64 Family 23 Model 113 Stepping 0, AuthenticAMD: 24 cores
Memory:        31.9 GB

mne:           0.22.1
numpy:         1.19.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.6.2
matplotlib:    3.3.4 {backend=Qt5Agg}

sklearn:       0.24.1
numba:         0.53.1
nibabel:       Not found
nilearn:       Not found
dipy:          Not found
cupy:          Not found
pandas:        1.2.3
mayavi:        Not found
pyvista:       Not found
vtk:           Not found
```

agramfort, 2021-04-14:
look at raw.first_samp attribue

>

skjerns, 2021-04-14:
thanks! that solves it :)

closed 2021-04-14 by skjerns

skjerns, 2021-04-14:
however, `raw.first_samp` is `51000`, but my delay is `5100`, in which unit is `first_samp` noted?

agramfort, 2021-04-14:
are you sure? first_samp is in samples not seconds

>

skjerns, 2021-04-14:
never mind, just a typo :D sorry!

I'm still a bit confused by how to use the event positions now when accessing arrays from `raw.get_data()`. I just always add `raw.first_samp` or retrieve the data by `raw.get_data(start=raw.first_samp)` to get fitting indices?

agramfort, 2021-04-14:
>
> I'm still a bit confused by how to use the event positions now when
> accessing arrays from raw.get_data().
>
with this first_samp trick you don't need to get new events if you crop the
raw object. We just update the first_samp.

> I just always add raw.first_samp or retrieve the data by
> raw.get_data(start=raw.first_samp) to get fitting indices?
>
the convention of get_data is to start from the first available sample so
you need to subtract first_samp from events[:, 0]
and then you can use the first column as start and stop in get_data

HTH

skjerns, 2021-04-15:
Is there a place where I can find more documentation on `first_samp`, and how it is used, or what its definition is?

I'm a bit puzzled because I have the case in multiple files that `events[-1,0] > len(raw)`. In my understanding the `fif` contains all samples, and `first_samp` tells `mne` at which position the actual data starts, right? So how can my events be out of bounds for `raw.get_data()`? As my epoching works fine, I assume there is some other underlying logic as well.

agramfort, 2021-04-15:
https://mne.tools/stable/glossary.html?highlight=first_samp

>