mne-tools/mne-python #9303: ENH: Adding a scale legend to plot_topo and other plotting functions

ramkpari, 2021-04-15:
#### Describe the new feature or enhancement
Plotting functions like `plot_topo` could use a scale legend to visually describe the exact scale of subplots being represented by each subplot. Something like this : 

[image]

The proposed `scale legend` showcases the X and Y scale used by each channel type in the initial plot and also the scale for each subplot on click. The parameter could also take an input to customize where the scale legend is located with inputs such as (`Left` and `Right`) or custom location parameters. 

This would potentially be a multi PR approach with the initial focus specifically for `plot_topo` and its different forms such as `plot_evoked_topo`, `plot_topo_image_epochs` etc. 

#### Describe your proposed implementation
One way of implementing this would be to add the Scale Legend as a separate subplot for each channel type with specifically mapped default locations. Since `plot_topo` currently only works with `unified=True`, this would be the easiest way to implement it. This does mean a new `Scale_Legend` class is created in the process. 

#### Describe possible alternatives
Another potential method is to work with a custom class created from `matplotlib.legend`. This does bring a unique set of complications based on how it interacts with `unified=True` inside `plot_topo` and the subplots as well. Theoretically, this method would be easier to maintain, although since it's still a custom child class it wouldn't differ by much.

Thoughts? @larsoner @agramfort @jasmainak @drammock @rob-luke 

References:
- [Example Image used above](https://aaltodoc.aalto.fi/bitstream/handle/123456789/15550/master_Jas_Mainak_2015.pdf) (Pg 35)

drammock, 2021-04-15:
note that this is already done automatically for `mne.viz.plot_compare_evokeds(evoked.pick('eeg'), axes='topo')` (see bottom right corner of plot).  See if it is easy to use the same helper function for both `plot_compare_evokeds` and `evoked.plot_topo` (and if not, maybe refactor it so that it would be possible).

ramkpari, 2021-04-16:
> note that this is already done automatically for `mne.viz.plot_compare_evokeds(evoked.pick('eeg'), axes='topo')` (see bottom right corner of plot). See if it is easy to use the same helper function [...]

Ahh, A little refactoring and extending the `_draw_legend_pce` to deal with 2 channel types instead of just one should be enough to add a scale legend to `evoked.plot_topo`

jasmainak, 2021-04-16:
Time for PR :)

rob-luke, 2021-04-22:
Great idea, thanks @ramkpari

larsoner, 2023-10-07:
FYI for now `plot_compare_evokeds(..., axes="topo")` can get you an axis in the lower right with a scale bar

https://mne.tools/dev/auto_tutorials/evoked/20_visualize_evoked.html#topographical-subplots