mne-tools/mne-python #9304: Removal of muscle artifacts

Mirrabella, 2021-04-16:
Hello, 
I have a very simple question, but I just can't find a clear answer for it. 
I am exercising to cleanse the muscle signal with the function annotate_muscle_zscore() .
As I understand, the function creates an annotation with which we can then mark the found muscle artifacts as "bad". And then, when we write the annotation to the raw data using the

raw.set_annotations (annot_muscle) function,

these tagged muscle artifact data segments will be automatically deleted when epochs are created. I understand correctly?

I was guided by this fragment (from the description of the function to mne.Annotations)
[image]

And what time interval of data will be cut when removing muscle artifacts? I did not really understand this point from the description of the function.:( 

Thank you in advance :)

hoechenberger, 2021-04-16:
Hello @Mirrabella!

> these tagged muscle artifact data segments will be automatically deleted when epochs are created. I understand correctly?

Yes, segments annotated with `bad_*` will be excluded when creating epochs.

> And what time interval of data will be cut when removing muscle artifacts?

I've never worked with this code myself, but after setting the annotations via `raw.set_annotations()`, you can visually inspect the data via `raw.plot()`; you will be able to see the annotated segments.

drammock, 2021-04-16:
usage questions are generally more appropriate on our forum (https://mne.discourse.group).  If there are any follow-up questions, please post them there.

closed 2021-04-16 by drammock