mne-tools/mne-python #9306: BUG: flat map with "both" on top of one another

kingjr, 2021-04-16:
Using `stc.plot(surface='flat', hemi='both')` leads to a superposition of the two hemispheres:

[image]

(in addition, the conventional orientation of flat map is rotated 90* compared to what mne plots:
[image]
)

larsoner, 2021-04-16:
1. We should just not allow `hemi='both'` when using a flatmap and tell people to use `hemi='split'` (ValueError)
2. We should fix the orientation, it's a regression since https://mne.tools/dev/auto_tutorials/source-modeling/plot_visualize_stc.html is wrong but https://mne.tools/stable/auto_tutorials/source-modeling/plot_visualize_stc.html is correct (can you confirm @kingjr ?)

larsoner, 2021-04-16:
@kingjr if you want to take a stab at the ori fix at least, it should be that this line needs an appropriate `roll=` in degrees:

https://github.com/mne-tools/mne-python/blob/main/mne/viz/_brain/view.py#L41

larsoner, 2021-04-16:
... although I guess we could allow `hemi='both'` and add an appropriate offset so they show up in the correct locations relative to one another. This might actually be preferable to `hemi='split'` in some situations. So I think I changed my mind on (1) above.

kingjr, 2021-04-16:
1. I vote for an offset, it's easier to handle zoom in some cases. 
2. hard to tell for an auditory response! I would only confirm for a visual one (alternatively, a quick check can be done by using the highlight by area option made by @GuillaumeFavelier)

it's hard for me to do PR on anything related to pyvista, because I only have it working on a headless, and I'm experiencing some painful lag

GuillaumeFavelier, 2021-04-16:
No worries, I can add that to my todo list and update https://github.com/mne-tools/mne-python/issues/7162

larsoner, 2021-04-16:
@GuillaumeFavelier can you take a look?

GuillaumeFavelier, 2021-04-16:
I'm on it :+1:

PR #9315 (2021-04-19): MRG, BUG: Fix positioning of flatmaps
Closes #9306
closed 2021-04-19 by drammock