mne-tools/mne-python #9307: add file like object support for raw ffile formats

rla3rd, 2021-04-16:
#### Describe the new feature or enhancement
*Please provide a clear and concise description of what you want to add or
change.

adding file lke object support would allow using mne in spark through the use of its binaryfile method
https://spark.apache.org/docs/latest/sql-data-sources-binaryFile.html  the idea would be to load up the edfs into a binary file dataframe and then do a map function using bytes as input.  this would allow distributed loading of edfs in spark across the cluster,

#### Describe your proposed implementation
*Describe how you think the feature or improvement should be implemented (e.g., 
as a new method on an existing class? as new capability added to an existing
method?) If you're not sure, please delete this section and the next section.*

the existing method could be overriden with a named parameter such as file_like = True

#### Describe possible alternatives
*If you've suggested an implementation above, list here any alternative
implementations you can think of, and brief comments explaining why the chosen
implementation is better.*

could also create a 2nd method signature just for file like objects

#### Additional comments
*Add any other context or screenshots about the feature request here.*

sappelhoff, 2021-06-28:
seems related: https://github.com/mne-tools/mne-python/issues/8976

larsoner, 2021-06-28:
Indeed closing as a duplicate of #8976

closed 2021-06-28 by larsoner