mne-tools/mne-python #9312: Cutting a new release

hoechenberger, 2021-04-18:
Hello all,

it's been 4 months since the release of MNE-Python 0.22.0, and we've had super busy contributors submitting a large stream of changess, producing a big list of enhancements and bug fixes:

https://mne.tools/dev/whats_new.html

Many of these will make the lives of our users more pleasant 😁 so I would suggest we should consider making a new release soon. The currently set due date of the 0.23 milestone is in approx. 4 weeks, and I believe we should aim to really meet the deadline this time.

What's your opinion?

larsoner, 2021-04-18:
Agreed we should have that deadline as an upper bound. I think folks should go through open issues and PRs and mark what needs to be done. I can probably make some bugfix PRs this week

agramfort, 2021-04-18:
+1 I don't know if we have current blockers. If not I don't see a reason to
wait more.

>

drammock, 2021-04-18:
there are several blockers I think.  Most prominent in my mind is that many of the 3D plots in the current dev docs have really messed up rendering (#9007 / #9010).  It would also be nice to get the PR that renames all the examples/tutorials finalized first (#9292), as well as the current website changes (#9288).

larsoner, 2021-04-19:
I went through and marked issues and PRs I see as blockers. I don't think it should take too long to take care of them

larsoner, 2021-04-26:
FYI based on the remaining 0.23-marked issues and PRs I plan to cut a release tomorrow

larsoner, 2021-04-27:
Closing, starting the process once CircleCI comes back green from the latest PR

closed 2021-04-27 by larsoner