mne-tools/mne-python #9319: Documentation for copy parameter in CSD inaccurate

fbaumgardt, 2021-04-19:
It appears that this line in the CSD documentation is not accurate:
https://github.com/mne-tools/mne-python/blob/93359566261be5c3fe7ceef8cb1fefce910cca52/mne/preprocessing/_csd.py#L88

I.e. I don't see any code for referencing based on the parameter. Also, the type validation happens after its use.

agramfort, 2021-04-19:
maybe for @alexrockhill or @dengemann ?

alexrockhill, 2021-04-19:
I think the note is for the code as it was previously and then the average reference got taken out. I don't think it should have, that was with mine and @larsoner's last PR fixing the CSD. I'll look at the old, not fixed version and see if the line that re-references should be put back in.

fbaumgardt, 2021-04-19:
Yeah, I'm just scanning through https://psychophysiology.cpmc.columbia.edu/Software/CSDtoolbox/faq.html to figure out if it's required. I had to remove an implicit reference but I'm still parsing out whether mastoid references are okay.

Thanks for the code and for dealing with the issue!

alexrockhill, 2021-04-19:
Hmm okay so maybe the note should be changed then...

alexrockhill, 2021-04-19:
Great, so I think the function works as is (besides the copy type validation but that doesn't really effect the output) but the documentation will be clarified when https://github.com/mne-tools/mne-python/pull/9320 merges and the validation will also be in the right order.

alexrockhill, 2021-04-20:
PR merged, I think this is good to be closed 👍

closed 2021-04-20 by larsoner