mne-tools/mne-python #9333: Kernel appears to have crashed with mne.viz

ilonadem, 2021-04-22:
MNE version 0.22.1
Python version 3.8.5
Remote jupyter notebook/google colab

**Describe the bug**

I have been trying to run several of the jupyter tutorials from the mne website in a remote jupyter notebook. Each time I try to run any command that uses mne.viz(), or any other interactive visualization, the kernel restarts and I get the message **"the kernel appears to have died"**

**Steps to Reproduce**

For example, I get the "kernel appears to have died" message when I run the following code:
```
%matplotlib inline

import mne
Brain = mne.viz.get_brain_class()

subjects_dir = mne.datasets.sample.data_path() + '/subjects'
mne.datasets.fetch_hcp_mmp_parcellation(subjects_dir=subjects_dir,
                                        verbose=True)

mne.datasets.fetch_aparc_sub_parcellation(subjects_dir=subjects_dir,
                                          verbose=True)

labels = mne.read_labels_from_annot(
    'fsaverage', 'HCPMMP1', 'lh', subjects_dir=subjects_dir)

brain = Brain('fsaverage', 'lh', 'inflated', subjects_dir=subjects_dir,
              cortex='low_contrast', background='white', size=(800, 600))
```
In particular, the kernel crashes on the last line, when there is a call to mne.viz.get_brain_class()

Is there a workaround (maybe specific to remote jupyter notebooks) to stop this crash from happening?

**Additional information**
I have tried running the same code in Google colab as well, which also results in the same problem.
I have also tried running other tutorials, which similarly crash whenever I try to run any of the mne.viz() functions

Platform: Remote jupyter notebook (in CoCalc), have also tried in Google Colab
Python: 3.8.5 
Executable: C:\Users\yxie\Anaconda3\python.exe
CPU: Intel(R) Xeon(R) CPU Model 85 Stepping 7
Memory: 32887760 kB

mne: 0.22.1
numpy: 1.19.5 {blas=D:\a\1\s\numpy\build\openblas_info, lapack=D:\a\1\s\numpy\build\openblas_lapack_info}
scipy: 1.5.0
matplotlib: 3.2.2 {backend=module://ipykernel.pylab.backend_inline}

sklearn: 0.0
numba: 0.51.2
nibabel: 3.1.0
nilearn: 0.6.2
dipy: Not found
cupy: Not found
pandas: 1.2.3
mayavi: Not found
pyvista: 0.29.1
vtk: 9.0.1

larsoner, 2021-04-22:
@GuillaumeFavelier it would be good if we could catch the no-OpenGL issue earlier, preferably in PyVista

In the meantime @ilonadem you could try our server environment on the remote machine to see if it works better:

https://github.com/mne-tools/mne-python/blob/main/server_environment.yml

Really what I think you need is some sort of suitably new OpenGL installed on the remote platform. MESA software rendering would work. I'm not sure about the best way to get this. The server_environment might already get it for you, not sure.

GuillaumeFavelier, 2021-04-22:
I will also suggest that you try:

```py
import mne
mne.viz.set_3d_backend("notebook")
...
```

To successfully enable the Notebook-specific 3d backend.

GuillaumeFavelier, 2021-04-22:
As a side note, Google Colab is not yet supported with this backend, you can follow the progress on this in the following thread:

https://github.com/mne-tools/mne-python/issues/8704#issue-781224584.

ilonadem, 2021-04-22:
Thank you for the quick response!

I have tried 
```
import mne
mne.viz.set_3d_backend("notebook")
```
which produces the same kernel crash, and already have OpenGL installed in the environment.

In the meantime, I'll go ahead and try using your your server environment

GuillaumeFavelier, 2021-04-23:
I'm trying to reproduce so I downloaded `sub-01_task-localizer_coherent+dSPM+hemi-lh.stc` and the `sub01` folder but I obtain the following when I run the script:

```
/tmp/bug.py in <module>
     10     subjects_dir = 'subjects'
     11     subject = 'sub-01'
---> 12     stc = mne.read_source_estimate(stc_fname)
     13 else:
     14     data_path = sample.data_path()

~/source/mne-python/mne/source_estimate.py in read_source_estimate(fname, subject)
    301             fname += '-stc'
    302         elif any(stc_exist) or any(w_exist):
--> 303             raise IOError("Hemisphere missing for %r" % fname_arg)
    304         else:
    305             raise IOError("SourceEstimate File(s) not found for: %r"

OSError: Hemisphere missing for 'sub-01_task-localizer_coherent+dSPM+hemi-lh.stc'
```

leosunpsy, 2022-11-21:
You can try the following operations：
(1) Make sure mesa is installed on your ubuntu：

sudo apt update && sudo apt upgrade -y
sudo apt install mesa-utils -y
glxinfo | grep "OpenGL version"

## Install/Upgrade Mesa Drivers – oibaf/graphics-drivers
sudo add-apt-repository ppa:oibaf/graphics-drivers -y
sudo apt update
sudo apt upgrade -y
glxinfo | grep "OpenGL version"
（2）Check the soft link " libstdc++.so.6" and the target file"libstdc++.so.6.0.30" (My system has libstdc++.so.6.0.30 in that folder, yours may be libstdc++.so.6.0.29 or 28)is the same (name and size) in both paths “/lib/x86_64-linux-gnu” and " /path to your anaconda/anaconda3/lib"

cd /lib/x86_64-linux-gnu
ls -al | grep libstdc++

cd /home/xxx/anaconda3/lib (I installed the anaconda here)
ls -al | grep libstdc++

If the above files are not the same in both paths：
（a）Delete the soft link “libstdc++.so, libstdc++.so.6” and target file “libstdc++.so.6.0.X” in the path “/home/xxx/anaconda3/lib”
（b）copy the target file “libstdc++.so.6.0.X” from “/lib/x86_64-linux-gnu” to " /home/xxx/anaconda3/lib":

sudo cp /lib/x86_64-linux-gnu/libstdc++.so.6.0.X  /home/xxx/anaconda3/lib
Create the soft links "libstdc++.so " and “libstdc++.so.6” for "libstdc++.so.6.0.X ":

  ln -s libstdc++.so.6.0.X libstdc++.so 
  ln -s libstdc++.so.6.0.X libstdc++.so.6
(3) Set up mne

mne.viz.set_3d_backend("pyvistaqt")
mne.viz.set_3d_options(antialias=False) 
False is useful when renderers have problems (such as software MESA renderers).

larsoner, 2023-10-01:
I think we have improved our error handling and instructions for MESA use nowadays. But let's reopen if things are still unusable for people

closed 2023-10-01 by larsoner