mne-tools/mne-python #9334: Consider making `raw.annotations` a property

cbrnr, 2021-04-22:
I would like to revive #5512. Every time I want to set annotations, I blindly assume that I can just assign to `raw.annotations`. However, MNE uses a `set_annotations` method, which is completely un-Pythonic IMO. I know the reason why @agramfort wanted to have this explicit setter (people won't expect that something happens on assignment, whereas a method will make that more explicit).

I still disagree. Properties are such a common thing in Python that people should be aware that setters implement some additional things (checks) before assignment. That's exactly why properties were created in the first place.

In addition, `raw.annotations` is *already* a property pointing to `raw._annotations`. Instead of `raw.set_annotations`, all we needed to do is change that to `@annotations.setter` and everything would fall into place.

agramfort, 2021-04-22:
just to repeat myself

with what you propose if you do:

raw.annotations = annotations
then you are very likely to get that
raw.annotations != annotations

it will be a copy and it can even be different (drop of annotations
outside + setting orig_time)

cbrnr, 2021-04-22:
Yes, but if this happens there will be at least info if not warning messages.

drammock, 2021-04-22:
> Properties are such a common thing in Python that people should be aware that setters implement some additional things (checks) before assignment.

anytime we say "people should be aware that..." I worry that we are raising the bar for how much Python experience people need before they can use MNE-Python effectively.  MNE-Python may be the first exposure to Python for many users, and for some it may even end up being the *only* domain in which they use Python. They will derive no advantage from this part of the API being more Pythonic, and they might encounter confusing pseudo-bugs because of it (i.e., the example @agramfort just gave).

cbrnr, 2021-04-22:
Hm. I overlooked the fact that a copy is being made. This is indeed something that does not happen in an assignment.

Why is the copy needed? Why can't users create an `Annotations` object and assign it the normal way?

agramfort, 2021-04-23:
>
> Why is the copy needed? Why can't users create an Annotations object and
> assign it the normal way?
>
to deal with the meas_date and drop annotations that do not fall inside the
data time range

larsoner, 2022-02-21:
Closing since I think we have converged on not changing this behavior

closed 2022-02-21 by larsoner