mne-tools/mne-python #9335: Show times relative to `meas_date` in `raw.plot`

cbrnr, 2021-04-22:
Based on my question in our forum (https://mne.discourse.group/t/show-times-relative-to-meas-date-in-raw-plot/3027/3), I think it would be useful to show time stamps based on `meas_date` on the x-axis of `raw.plot`. This should be optional, and by default `raw.plot` should stick to showing time starting at 0.

larsoner, 2021-04-26:
In terms of a mechanism for doing it, I would make a new tick formatter. Matplotlib decides where the ticks go, and the formatter takes the x values (which are just from `raw.times` starting from zero) and incorporates whatever information about `first_samp`, `raw.annotations.meas_date` etc. and returns the datetime string.

cbrnr, 2021-04-27:
I can give it a try, but not for 0.23 (obviously).

cbrnr, 2021-06-09:
Fixed in #9419.

closed 2021-06-09 by cbrnr