mne-tools/mne-python #9338: Add support for NIRx fNIRS data in SNIRF format

rob-luke, 2021-04-23:
#### Describe the new feature or enhancement
We currently have a SNIRF reader that works with files produced using the society for fNIRS approved generator/converter.

However, there is quite some flexibility in the specification. Further, we only support a subset of the specifiation (for example we do not support anything other than continuous wave spectroscopy, and we dont support 2D locations).

NIRx now export their data as SNIRF. This is great. However, they do not export some fields that are optional in the spec. And currently our reader expects these data fields. So the reader needs to be modified to be forgiving if these optional fields are missing, or it needs to throw a usefull error.

#### Describe your proposed implementation

I have recorded some data with a NIRScout2 device in snirf format. I will add this to the test data repository and I will then update our reader to gracefully handle these missing fields.

My initial investigation indicates that the only missing fields are
* /nirs(i)/probe/sourceLabels
* /nirs(i)/probe/detectorLabels

so this shouldnt be too much work for me.

I will also use this oppurtunity to all other fields we expect are "required" and try not to rely on optional fields.

#### Additional comments

I am writing this here so everyone knows and does not try to duplicate effort.

I will also address https://github.com/mne-tools/mne-nirs/issues/257

This will not address https://github.com/mne-tools/mne-python/issues/9308

FYI @robquant @HanBnrd @larsoner

rob-luke, 2021-04-23:
In the case that these fields are missing I am going to assign the following naming structure...

* Sources: S1, S2, S3, etc
* Detectors: D1, D2, D3, etc

larsoner, 2021-04-23:
All sounds good @rob-luke !

PR #9347 (2021-04-26): MRG: Add SNIRF support for 2D locations and NIRx files
#### Reference issue
Closes #9338 mne-tools/mne-nirs#257 #9308.

#### What does this implement/fix?
This PR enables reading of SNIRF files from the NIRx vendor and it also enables reading of SNIRF files with 2D locations from the SfNIRS team.

Support for 2D locations was enabled by setting the z coordinate to 0.

Support for NIRx was enabled by not relying on the `sourceLabels` field in SNIRF. Which was optional in the spec, so I shouldn't have expected it to be there.

#### Additional information
People may wish to support no sensor loca [... cut]
closed 2021-04-26 by larsoner