mne-tools/mne-python #9339: raw.plot raises KeyError with event_colors

cbrnr, 2021-04-23:
Some code I wrote a while ago suddenly started to throw a `KeyError`:

```python
import numpy as np
import mne

fs = 512
raw = mne.io.RawArray(np.random.randn(10, 10 * fs) * 1e-6, mne.create_info(10, fs, "eeg"))
events = np.array([[1 * fs, 0, 1], [2 * fs, 0, 2], [3 * fs, 0, 3]])

raw.plot(events=events, event_color={1: "red", 2: "blue"})
```

@drammock do you know if this behavior changed? Should we assume a default color for events that are not specified in the `event_color` dict?

drammock, 2021-04-26:
This is expected behavior, and is documented.  See bolded section of docstring below:

```
event_color : color object | dict | None
    Color(s) to use for events. To show all events in the same color, pass any
    matplotlib-compatible color. To color events differently, pass a `dict`
    that maps event names or integer event numbers to colors **(must include
    entries for *all* events, or include a "fallback" entry with key ``-1``).**
    If ``None``, colors are chosen from the current Matplotlib color cycle.
    Defaults to ``'cyan'``.
```

When the `event_color` / `event_colors` args of raw and epochs were unified, doing it this way was as close to maintaining backwards-compatibility as I could manage (see item 5 [here](https://github.com/mne-tools/mne-python/pull/8381#issuecomment-722720232)).

> Should we assume a default color for events that are not specified in the `event_color` dict?

Not in my opinion.  How would we pick that color?  What would we do if that color (or something very visually close to it) were already in the user-supplied dict?  If there's a fallback, should it be the same for all events not in the dict, or a unique fallback for each?  I don't think there's a principled answer to these questions.

closed 2021-04-26 by drammock

cbrnr, 2021-04-27:
I didn't see the new docstring, and I can live with explicitly passing a `-1` dict entry.