mne-tools/mne-python #9343: Trouble converting an Annotations object to an Events array ('Annotations' object has no attribute 'annotations')

annnhuang, 2021-04-23:
I am a beginner to Python MNE. I got super stuck on trying to convert the annotations into events...

**The following are the codes I have:**

``# reading embedded events as annotations

raw_annot = mne.read_annotations('C:\\Users\\HIVE-Admin\\Desktop\\Ann_EEG\\EEG_results\\s01.vmrk')
raw.set_annotations(events)
print(events)

##convert an Annotations object to an Events array

event_id = {'wrong response': 81,
              'wrong response': 82,
              'correct response': 91,
              'correct response': 92,
              'learning/target/right': 100,
              'learning/target/left': 101,
              'learning/old/target/right': 102,
              'learning/old/target/left': 103,
              'testing/target/right': 200,
              'testing/target/left': 201,
              'testing/old/target/right': 202,
              'testing/old/target/left': 203}

events_id = mne.events_from_annotations(raw_annot, event_id= event_id)´´

**I keep getting the following error:**

Finding 'sfreq' from header file: C:\Users\HIVE-Admin\Desktop\Ann_EEG\EEG_results\s01.vhdr
<Annotations | 3930 segments: Comment/ControlBox is not connected via USB ...>

AttributeError                            Traceback (most recent call last)
<ipython-input-19-054753c0b8d4> in <module>
     22               'testing/old/target/left': 203}
     23 
---> 24 events_id = mne.events_from_annotations(raw_annot, event_id= event_id)

<decorator-gen-50> in events_from_annotations(raw, event_id, regexp, use_rounding, chunk_duration, verbose)

~\anaconda3\lib\site-packages\mne\annotations.py in events_from_annotations(raw, event_id, regexp, use_rounding, chunk_duration, verbose)
   1011     integer event code ``1``, ``'2'`` to ``2``, etc.
   1012     """
-> 1013     if len(raw.annotations) == 0:
   1014         event_id = dict() if not isinstance(event_id, dict) else event_id
   1015         return np.empty((0, 3), dtype=int), event_id

AttributeError: 'Annotations' object has no attribute 'annotations'

cbrnr, 2021-04-23:
You need to pass the `raw` object instead of `raw_annot`, i.e.

```
mne.events_from_annotations(raw, event_id= event_id)
```

annnhuang, 2021-04-23:
I did that, but I got the following error, (one that I also keep encountering):

ValueError: Could not find any of the events you specified.

cbrnr, 2021-04-23:
Then `raw.annotations` doesn't contain annotations with the given type. Is your initial code correct? I think the line

```python
raw.set_annotations(events)
```

should be

```python
raw.set_annotations(raw_annot)
```

Then you can check which annotations you have by printing `raw.annotations`.