mne-tools/mne-python #9350: `mne.time_frequency.multitaper` Nyquist adjustment slightly incorrect

ericphanson, 2021-04-26:
In https://github.com/mne-tools/mne-python/blob/c6b22e82b64da0ed0f52a5f28bd4711e47090f9e/mne/time_frequency/multitaper.py#L322

there’s an adjustment of the Nyquist frequency component if the length of the signal input is even. I think the condition should instead be if `n_fft` is even. I think the reasoning is the same as in https://www.mathworks.com/matlabcentral/answers/102727-why-does-the-example-in-fft-divide-the-fft-result-by-the-length-of-the-signal-and-multiply-the-magni#answer_112072.

Originally mentioned in https://mne.discourse.group/t/understanding-mne-time-frequency-csd-multitaper-implementation

larsoner, 2021-04-26:
@ericphanson that sounds right to me, are you up for making a PR to fix it?

ericphanson, 2021-04-26:
Sorry, I'm not very familiar with Python and I suspect that a lot of tests will need to be updated, so I'd prefer to just make the report.

cgohil8, 2022-08-23:
I'm happy to work on this.

cgohil8, 2022-08-23:
This was fixed in https://github.com/mne-tools/mne-python/pull/10541.

closed 2022-08-23 by drammock