mne-tools/mne-python #9360: Mayavi problems on MAC

bloyl, 2021-04-28:
I'm having problems with the macos and mayavi visulaization in mne 0.23 (but probably earlier versions as well).

<details>
<summary markdown="span">MAC hardware</summary>

  ``` 
  macOS Catalina 
  Version 10.15.7
    Model Name:	MacBook Pro
    Model Identifier:	MacBookPro15,2
    Processor Name:	Quad-Core Intel Core i7
    Processor Speed:	2.8 GHz

   Intel Iris Plus Graphics 655:
      Chipset Model:	Intel Iris Plus Graphics 655
      Type:	GPU
   ```
</details>

I created a fresh environment for mne 0.23

<details>
<summary markdown="span">Make env</summary>

```
 conda env remove -n mne-stable
 curl --remote-name https://raw.githubusercontent.com/mne-tools/mne-python/master/environment.yml
 conda env create -n mne-stable --file environment.yml
 conda activate mne-stable
```
</details>

<details>
<summary markdown="span">mne sys_info</summary>

```
(mne-stable) bloyl@507b9dedf1fe ~ % mne sys_info
Platform:      macOS-10.15.7-x86_64-i386-64bit
Python:        3.9.2 | packaged by conda-forge | (default, Feb 21 2021, 05:02:20)  [Clang 11.0.1 ]
Executable:    /Users/bloyl/anaconda3/envs/mne-stable/bin/python3.9
CPU:           i386: 8 cores
Memory:        16.0 GB

mne:           0.23.0
numpy:         1.20.2 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:         1.6.3
matplotlib:    3.4.1 {backend=MacOSX}

sklearn:       0.24.1
numba:         0.53.1
nibabel:       3.2.1
nilearn:       0.7.1
dipy:          1.4.0
cupy:          Not found
pandas:        1.2.4
mayavi:        4.7.2
pyvista:       0.29.1 {pyvistaqt=0.3.0, OpenGL 4.1 INTEL-14.7.8 via Intel(R) Iris(TM) Plus Graphics 655}
vtk:           9.0.1
PyQt5:         5.12.3
```
</details>

The pyvista backend seems to work, however the mayavi backend just shows grey background plots.

### To reproduce:
```
(mne-stable) bloyl@507b9dedf1fe ~ % export MNE_3D_BACKEND=mayavi; ipython
Python 3.9.2 | packaged by conda-forge | (default, Feb 21 2021, 05:02:20)
Type 'copyright', 'credits' or 'license' for more information
IPython 7.22.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import mne
   ...: from mne.viz import plot_alignment, set_3d_view
   ...:
   ...: print(__doc__)
   ...:
   ...: data_path = mne.datasets.sample.data_path()
   ...: subjects_dir = data_path + '/subjects'
   ...: trans = mne.read_trans(data_path + '/MEG/sample/sample_audvis_raw-trans.fif')
   ...: raw = mne.io.read_raw_fif(data_path + '/MEG/sample/sample_audvis_raw.fif')
   ...: # Plot electrode locations on scalp
   ...: fig = plot_alignment(raw.info, trans, subject='sample', dig=False,
   ...:                      eeg=['original', 'projected'], meg=[],
   ...:                      coord_frame='head', subjects_dir=subjects_dir)
   ...: # Set viewing angle
   ...: set_3d_view(figure=fig, azimuth=135, elevation=80)
Automatically created module for IPython interactive environment
Opening raw data file /Users/bloyl/mne_data/MNE-sample-data/MEG/sample/sample_audvis_raw.fif...
    Read a total of 3 projection items:
        PCA-v1 (1 x 102)  idle
        PCA-v2 (1 x 102)  idle
        PCA-v3 (1 x 102)  idle
    Range : 25800 ... 192599 =     42.956 ...   320.670 secs
Ready.
Using outer_skin.surf for head surface.
Using mayavi 3d backend.

qt.qpa.drawing: Layer-backing can not be explicitly controlled on 10.14 when built against the 10.14 SDK

In [2]:
```

it will generate a grey window ('TVTK Scene') with no mayavi control bar. I don't think the QT warning is the issue as it also displays with the pyvista backend.

Interestingly if I press `cntl-d` to quit ipython. the figure shows up while ipython asks `Do you really want to exit ([y]/n)?`
If I press no, the plot remains but the window is non-responsive.

I should note that mayavi plotting functions (https://docs.enthought.com/mayavi/mayavi/auto/mlab_helper_functions.html#mayavi.mlab.contour3d) seem to work fine.

The reason this is an issue is that `mne coreg` is still mayavi based and displays the same problem.

larsoner, 2021-04-29:
I can't replicate on my macOS system with Intel Iris graphics.

Does `python -ic "from mayavi import mlab; mlab.test_plot3d()"` work where it displays and you can interact?

What about `MPLBACKEND=Qt5Agg python -ic "import matplotlib.pyplot as plt; plt.ion(); plt.plot(0, 0)"` where you can interact and zoom?

bloyl, 2021-04-29:
Both 
`python -ic "from mayavi import mlab; mlab.test_plot3d()"`
and
`MPLBACKEND=Qt5Agg python -ic "import matplotlib.pyplot as plt; plt.ion(); plt.plot(0, 0)"`
work fine.

note 2d plotting (matplotlib) is working fine. I was able to run this example without issue (https://mne.tools/stable/auto_examples/visualization/channel_epochs_image.html#sphx-glr-auto-examples-visualization-channel-epochs-image-py)

similarly 3d plotting (pyvista) also seems to  be fine. its only the `mayavi` backend

larsoner, 2021-04-29:
> Both ... and ... work fine... similarly 3d plotting (pyvista) also seems to be fine. its only the mayavi backend

Okay but it's not a purely Mayavi problem because the `mlab.test_plot3d()` worked, in the sense that it plotted something like this that you could rotate with the mouse, etc., right?

[image]

In other words, Mayavi itself is fine, but somehow our use of Mayavi / the mayavi backend is broken.

Next can you try:
```
$ python -ic "import mne; from surfer import Brain; brain = Brain('fsaverage', 'lh', 'inflated', subjects_dir=mne.datasets.sample.data_path() + '/subjects')"
```
You might need to `pip install pysurfer`.

It just occurred to me that this might be an antialiasing issue since it's broken on most macs. Can you see if setting `MNE_3D_OPTION_ANTIALIAS=false` as an env var helps?

Your `sys.platform` is `'darwin'`, right?

bloyl, 2021-04-29:
`MNE_3D_OPTION_ANTIALIAS=false` solves the problem with `mne coreg` which was the real show stopped for me.

However there is still something strange going on.

the following behavior isn't changed by `MNE_3D_OPTION_ANTIALIAS=false` and to confirm `sys.platform` is 'darwin'

`python -ic "import mne; from surfer import Brain; brain = Brain('fsaverage', 'lh', 'inflated', subjects_dir=mne.datasets.sample.data_path() + '/subjects')"`
works just fine.

but running the same thing in `ipython` does this:

https://user-images.githubusercontent.com/5585821/116600297-279e3400-a8f7-11eb-8535-6ce31171be4d.mov

note similar behavior occurs if I run 

`export MNE_3D_BACKEND=mayavi; ipython -i ~/Projects/mne-python/mne-python-devel/examples/visualization/plot_eeg_on_scalp.py`

which works with python calls.

;---
Since `mne coreg` works (with `MNE_3D_OPTION_ANTIALIAS=false`) and I generally use the `pyvista` backend feel free to close if no-one else can replicate.

larsoner, 2021-04-29:
> MNE_3D_OPTION_ANTIALIAS=false solves the problem with mne coreg which was the real show stopped for me.

Great! We should add this to the 3D debugging docs. @GuillaumeFavelier want to take a stab?

> but running the same thing in ipython does this:

This looks like an ipython<->mayavi bug. I wonder if it would be fixed by doing `%gui qt` or `%matplotlib qt`

bloyl, 2021-04-29:
`MNE_3D_OPTION_ANTIALIAS` is on the advanced setup page.

but I only tested in the ipython session where it didn't have an effect. so never tested it against `mne coreg`

Its up to you 2 if you feel the current doc is sufficient.

larsoner, 2021-04-29:
@bloyl I would argue that if it tripped you up, it's probably not sufficient :)

And looking at it, this is listed in the `Off-screen rendering with MESA` section, not the `Troubleshooting` section. I think it should live there as well, and probably also specifically mention macOS as possibly needing this parameter.

closed 2021-04-30 by agramfort