mne-tools/mne-python #9370: maxfiltered split files not handled correctly (mne-0.23)

RuKrei, 2021-05-03:
#### Describe the bug
Dear mne developers,

as you all know, elekta (megin) fif files are automatically split, if file size exceeds 2GB.
This results in a second file with tag "-1".fif.
Mne handles this correctly.

However, if files are processed with maxfilter (in my case tsss + head transposition), by default resulting files are renamed to  
 - filename + maxfilter-tags .fif

If a split-file is processed, renaming results in:
 - filename + "-1" + maxfilter-tags .fif
-------------

#### Now here is the bug:
When loading maxfiltered split files, mne expects them to be named 
 - file + maxfilter-tags + "-1".fif

instead of 
 - file + "-1" + maxfilter-tags .fif

#### Here is the error message:
ValueError: Split raw file detected but next file .../RS10021988_block2_trans_tsss-1.fif does not exist. Ensure all files were transferred properly and that split and original files were not manually renamed on disk (split files should be renamed by loading and re-saving with MNE-Python to preserve proper filename linkage).

--> The name of the correct file would be RS10021998_block2-1_trans_tsss.fif

Thank you for fixing this.

Best
Rudi K.

#### Additional information
Platform:      Linux-5.8.0-50-generic-x86_64-with-glibc2.10
Python:        3.8.5 (default, Aug  5 2020, 08:36:46)  [GCC 7.3.0]
Executable:    /home/idrael/anaconda3/envs/mne_023/bin/python
CPU:           x86_64: 36 cores
Memory:        125.5 GB

mne:           0.23.0
numpy:         1.19.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.5.0
matplotlib:    3.2.2 {backend=Qt5Agg}

sklearn:       0.23.1
numba:         0.50.1
nibabel:       3.1.1
nilearn:       0.6.2
dipy:          1.1.1
cupy:          Not found
pandas:        1.1.0
mayavi:        4.7.2.dev0
pyvista:       0.25.3 {pyvistaqt=0.1.1, OpenGL 3.3 (Core Profile) Mesa 20.0.8 via llvmpipe (LLVM 10.0.0, 256 bits)}
vtk:           9.0.1
PyQt5:         5.15.0

agramfort, 2021-05-03:
the name of the next fif file is written in the file. I suspect your files
were renamed manually.

can you confirm?

>

RuKrei, 2021-05-11:
...sorry, yes - files have been renamed before I got them.
closing the Issue.

closed 2021-05-11 by RuKrei