mne-tools/mne-python #9373: Morphing STC is broken/ issue with STC plot

dasdiptyajit, 2021-05-05:
#### Describe the bug
It seems like there is some sort of bug when I try to compute : compute_source_morph to morph individual subject to fsaverage 
and STC.plot is also causing some issues.

#### Steps to reproduce (code)
_**# read the inverse
inv_fname = join(subjects_dir, id, int_name + inv_pattern)
inv = read_inverse_operator(inv_fname)

# source localization
stc = apply_inverse(evoked, inv, lambda2=lambda2, method=method, pick_ori='normal')
morph = compute_source_morph(src=inv['src'], subject_from=id, subject_to='fsaverage',
                             subjects_dir=subjects_dir, spacing=5)
stc_fsaverage = morph.apply(stc)
surfer_kwargs = dict(hemi='both', subjects_dir=subjects_dir, time_unit='s',
                     size=(800, 800), time_viewer=True)
brain = stc.plot(**surfer_kwargs)**_

#### Expected results
*Replace this text with a description of what you expected to happen.*
Morphing to fsaverage atlas 

#### Actual results

**Error while morphing:**
.
.
.
[done]
Applying inverse operator to "dev"...
    Picked 184 channels from the data
    Computing inverse...
    Eigenleads need to be weighted ...
    Computing residual...
    Explained  59.5% variance
    dSPM...
[done]
Writing STC to disk...
[done]
**Traceback (most recent call last):**

  File "/media/diptyajit/Ext_drive/Pycharm_Aud_Cog/P3_source_localizytion_script_global_main.py", line 708, in <module>
    source_localization(subject_ids=subject_ids, subjects_dir=subjects_dir, method=method,

  File "/media/diptyajit/Ext_drive/Pycharm_Aud_Cog/P3_source_localizytion_script_global_main.py", line 183, in source_localization
    morph = compute_source_morph(src=inv['src'], subject_from=id, subject_to='fsaverage',

  File "<decorator-gen-327>", line 22, in compute_source_morph

  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.9/site-packages/mne/morph.py", line 248, in compute_source_morph
    morph_mat = _compute_morph_matrix(

  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.9/site-packages/mne/morph.py", line 1112, in _compute_morph_matrix
    morpher.append(_hemi_morph(

  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.9/site-packages/mne/morph.py", line 1149, in _hemi_morph
    mm = _surf_upsampling_mat(vertices_from, e, smooth, warn=warn)

  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.9/site-packages/mne/morph.py", line 1287, in _surf_upsampling_mat
    use_e = e[:, idx] if len(idx) < n_tot else e

  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.9/site-packages/scipy/sparse/_index.py", line 33, in __getitem__
    row, col = self._validate_indices(key)

  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.9/site-packages/scipy/sparse/_index.py", line 147, in _validate_indices
    col = self._asindices(col, N)

  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.9/site-packages/scipy/sparse/_index.py", line 170, in _asindices
    raise IndexError('index (%d) out of range' % max_indx)

IndexError: index (146475) out of range

**Error while plot only subject Subject STC:**
Using pyvista 3d backend.

Using control points [ 7.33885576  9.56166922 26.24656785]
Traceback (most recent call last):

  File "/media/diptyajit/Ext_drive/Pycharm_Aud_Cog/P3_source_localizytion_script_global_main.py", line 708, in <module>
    source_localization(subject_ids=subject_ids, subjects_dir=subjects_dir, method=method,

  File "/media/diptyajit/Ext_drive/Pycharm_Aud_Cog/P3_source_localizytion_script_global_main.py", line 188, in source_localization
    brain = stc.plot(**surfer_kwargs)

  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.9/site-packages/mne/source_estimate.py", line 651, in plot
    brain = plot_source_estimates(

  File "<decorator-gen-153>", line 24, in plot_source_estimates

  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.9/site-packages/mne/viz/_3d.py", line 1780, in plot_source_estimates
    return _plot_stc(

  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.9/site-packages/mne/viz/_3d.py", line 1919, in _plot_stc
    brain.add_data(**use_kwargs)

  File "<decorator-gen-173>", line 22, in add_data

  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.9/site-packages/mne/viz/_brain/_brain.py", line 1952, in add_data
    self.set_data_smoothing(self._data['smoothing_steps'])

  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.9/site-packages/mne/viz/_brain/_brain.py", line 2746, in set_data_smoothing
    smooth_mat = _hemi_morph(

  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.9/site-packages/mne/morph.py", line 1149, in _hemi_morph
    mm = _surf_upsampling_mat(vertices_from, e, smooth, warn=warn)

  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.9/site-packages/mne/morph.py", line 1287, in _surf_upsampling_mat
    use_e = e[:, idx] if len(idx) < n_tot else e

  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.9/site-packages/scipy/sparse/_index.py", line 33, in __getitem__
    row, col = self._validate_indices(key)

  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.9/site-packages/scipy/sparse/_index.py", line 147, in _validate_indices
    col = self._asindices(col, N)

  File "/home/diptyajit/anaconda3/envs/mne/lib/python3.9/site-packages/scipy/sparse/_index.py", line 170, in _asindices
    raise IndexError('index (%d) out of range' % max_indx)

IndexError: index (146475) out of range

#### Additional information
*Replace this text with the output of `mne.sys_info()`.*
Platform:      Linux-4.15.0-132-generic-x86_64-with-glibc2.27
Python:        3.9.2 | packaged by conda-forge | (default, Feb 21 2021, 05:02:46)  [GCC 9.3.0]
Executable:    /home/diptyajit/anaconda3/envs/mne/bin/python
CPU:           x86_64: 4 cores
Memory:        15.6 GB

mne:           0.23.0
numpy:         1.20.2 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:         1.6.3
matplotlib:    3.4.1 {backend=Qt5Agg}

sklearn:       0.24.2
numba:         0.53.1
nibabel:       3.2.1
nilearn:       0.7.1
dipy:          1.4.0
cupy:          Not found
pandas:        1.2.4
mayavi:        4.7.2
pyvista:       0.29.1 {pyvistaqt=0.3.0, OpenGL 3.3 (Core Profile) Mesa 19.2.8 via Mesa DRI Intel(R) HD Graphics 530 (Skylake GT2)}
vtk:           9.0.1
PyQt5:         5.12.3

dasdiptyajit, 2021-05-05:
@larsoner @agramfort any idea what it could be? 
I tried to open the stc file with mne-C but it doesn't load and I am getting an error: **Overlay doesn't match the loaded surfaces**

agramfort, 2021-05-06:
are you sure there is no mistake ? wrong subject in inverse or wrong
freesurfer recon?

>

dasdiptyajit, 2021-05-06:
@agramfort I tested it yesterday and can confirm, its not a mne issue. Apparently, I was running fsfast pipeline for my fMRI data and for some subjects, some of the surfaces were missing or not updated since they were created with a older version of freesurfer recon. So, I updated the recon and it modified the coordinate no. in (lh/rh.white surfaces) and now the morhping doesn't working on the fly and I get the above error. I would certainly prefer to keep my old analysis files but then I can't continue since I need to do some further analysis for my MEEG data that utilize the morphing. Is any way to reconstruct a morph matrix from computed 'morphed stc' to stc files between a subject?? If I can reconstruct the morphed matrix then I can use it for the further analysis

agramfort, 2021-05-06:
if you recomputed the recon you'll need to recompute the src, fwd, inv etc.

dasdiptyajit, 2021-05-06:
Hi! Is it a must? I managed to get  the back up of the earlier version of recon surfaces and It works now. What's your recommendation?

agramfort, 2021-05-06:
if you have your old files then you're ok.

please close the issue if you conclude it's not an MNE issue.

>

dasdiptyajit, 2021-05-06:
thanks! I was about to do that

closed 2021-05-06 by dasdiptyajit